sabamdarif/termux-desktop #5: Please help 🙏 black screen on termux x11 and not working properly in vnc viewer also

Aditya7069, 2024-05-24:


sabamdarif, 2024-05-24:
first do nano $HOME/.vnc/xstartup and check if this "startxfce4 &" avilable or not if not then do echo "startxfce4 &" > $HOME/.vnc/xstartup chmod +x $HOME/.vnc/xstartup

if still don't work so do this 

apt reinstall xfce4 xfce4-goodies -y
apt reinstall termux-x11-nightly tigervnc -y

or 
do nano start.sh and pest this 
 
XDG_RUNTIME_DIR=${TMPDIR} termux-x11 :1.0 &
sleep 1
am start --user 0 -n com.termux.x11/com.termux.x11.MainActivity > /dev/null 2>&1
sleep 1
env DISPLAY=:1.0  dbus-launch --exit-with-session xfce4-session & 

type bash start.sh  and send me a  screenshot of the terminal or the send me the full message it print in the terminal

Aditya7069, 2024-05-25:
Issue was fix when I written vncserver

Aditya7069, 2024-05-25:
Thank you for telling I publish on YouTube channel Aditya virtual Android
master thanks

sabamdarif, 2024-05-25:
waiting  for your video 👍

closed 2024-05-25 by sabamdarif

Aditya7069, 2024-05-25:
It's working!

Aditya7069, 2024-05-25:
But have sound problem can you tell me how to fix in chromium

sabamdarif, 2024-05-25:
> […](#)

Give me some time I have to check i always use Firefox so I so don't know 

I will recommend you just reinstall pulseaudio

sabamdarif, 2024-05-27:
for now see this it's a chromium problem i will add it soon
https://github.com/termux/termux-app/issues/2554