sabamdarif/termux-desktop #6: New user isn't existed in proot-distro's /etc/passwd & Running commands in proot-distro's terminal shows cannot found libld.so

MCredbear, 2024-06-08:
I don't know if this is normal, and I open proot-distro's terminal by `debian` , `tx11start`, then open terminal in termux-x11

sabamdarif, 2024-06-08:
where did you type the tx11start command inside debian or inside termux ?

sabamdarif, 2024-06-08:
and for libld.so , i think it's problem of debian installation the debian (proot distro) installation might corrupted in the process

MCredbear, 2024-06-08:
> where did you type the tx11start command inside debian or inside termux ?

in proot debian

MCredbear, 2024-06-08:
> and for libld.so , i think it's problem of debian installation the debian (proot distro) installation might corrupted in the process

it is strange that, just run commands after `debian` in proot CLI works fine, but in the desktop terminal after `tx11start` in proot CLI, this occurs

sabamdarif, 2024-06-09:
> tx11start

the `tx11start` is for termux itself not for debian

MCredbear, 2024-06-09:
> > tx11start
> 
> the `tx11start` is for termux itself not for debian

So the correct way to run proot app is to run `pdrun [COMMAND]` in termux itself's desktop's terminal? Termux and proot sharing the same `bin` and `$PATH` confuses me....

sabamdarif, 2024-06-09:
Yes

sabamdarif, 2024-06-09:
for the _/etc/passwd_ issue , login into debian `pd login debian`  and type `passwd username`

MCredbear, 2024-06-09:
> for the _/etc/passwd_ issue , login into debian `pd login debian` and type `passwd username`

I typed `adduser username` after logined which also fixes it, but no idea whether it causes new issues

sabamdarif, 2024-06-09:
what issue ?

MCredbear, 2024-06-09:
> what issue ?

currently nope

MCredbear, 2024-06-09:
My SoC is Qcom 8gen1 , I chose turnip driver for proot debian, but when I run `pdrun glxgears -info`, it shows llvmpip driver, then I run `pdrun sudo apt install mesa-utils` and rerun `pdrun glxgears -info`, it it shows 
```
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
libGL error: glx failed to create drisw screen
libGL error: failed to load driver: zink
Error: glXCreateContext failed
```

MCredbear, 2024-06-09:
Another question: If I want to open a proot directory in code-oss, what should I do?

sabamdarif, 2024-06-09:
> pdrun sudo apt install mesa-utils

don't use `pdrun sudo apt install mesa-utils` insted use `debian install mesa-utils`
i think you didn't read all documentation

and for error: XDG_RUNTIME_DIR is invalid or not set in the environment i have to look for it

sabamdarif, 2024-06-09:
> Another question: If I want to open a proot directory in code-oss, what should I do?

> Another question: If I want to open a proot directory in code-oss, what should I do?

the Desktop, Music, Downloads, Pictures and Videos folders of the proot-distro are link with the termux home directory's folders

sabamdarif, 2024-06-09:
> the Desktop, Music, Downloads, Pictures and Videos folders of the proot-distro are link with the termux home directory's folders

if you create any new folder in proot-distro's home directory if have open it manuanny but i will add the option soon

MCredbear, 2024-06-11:
> > pdrun sudo apt install mesa-utils
> 
> don't use `pdrun sudo apt install mesa-utils` insted use `debian install mesa-utils` i think you didn't read all documentation

sorry, but
[image]

sabamdarif, 2024-06-11:
It can't handle .deb file now

And also vs code natively available it termux

MCredbear, 2024-06-11:
> It can't handle .deb file now
> 
> And also vs code natively available it termux

But how can I use some vscode plugins that require toolchains in proot, such as gdb and dart, these plugins need to launch the binaries in proot

sabamdarif, 2024-06-11:
> > It can't handle .deb file now
> > And also vs code natively available it termux
> 
> But how can I use some vscode plugins that require toolchains in proot, such as gdb and dart, these plugins need to launch the binaries in proot

first login into debian
then run this on debian
`curl https://packages.microsoft.com/keys/microsoft.asc | gpg --dearmor > microsoft.gpg`
`sudo install -o root -g root -m 644 microsoft.gpg /etc/apt/keyrings/microsoft-archive-keyring.gpg`
`sudo sh -c 'echo "deb [arch=amd64,arm64,armhf signed-by=/etc/apt/keyrings/microsoft-archive-keyring.gpg] https://packages.microsoft.com/repos/vscode stable main" > /etc/apt/sources.list.d/vscode.list'`

`sudo apt-get update`

logout from debian

in termux type `debian install code`

i think it will help you to install vscode

sabamdarif, 2024-06-11:
First login into debian then run

`sudo apt update`
`sudo apt install gnupg2 software-properties-common apt-transport-https curl`
`curl -sSL https://packages.microsoft.com/keys/microsoft.asc | sudo apt-key add -`
`sudo add-apt-repository "deb [arch=arm64] https://packages.microsoft.com/repos/vscode stable main"
`
change the `arm64` according to your system like:- `amd64`,`armhf`

`sudo apt update`

logout from debian

`debian install code`

i think it will help you to install vscode

sabamdarif, 2024-06-11:
also for ls command install lsd in debian

MCredbear, 2024-06-13:
> First login into debian then run
>
> `sudo apt update` `sudo apt install gnupg2 software-properties-common apt-transport-https curl` `curl -sSL https://packages.microsoft.com/keys/microsoft.asc | sudo [...]

OK, succeeded to install and add it to menu, but no luck to run it, sad

sabamdarif, 2024-06-13:
> OK, succeeded to install and add it to menu, but no luck to run it, sad

type `pdrun code` and see what's the error

MCredbear, 2024-06-13:
> type `pdrun code` and see what's the error

[image]
it detached the shell

sabamdarif, 2024-06-13:
i think vscode isn't installed , login into debian and run `code` command

MCredbear, 2024-06-13:
> i think vscode isn't installed , login into debian and run `code` command

[image]

sabamdarif, 2024-06-13:
check if `xorg-xhost` installed or not in termux

MCredbear, 2024-06-14:
> check if `xorg-xhost` installed or not in termux

[image]
Yes, bro

sabamdarif, 2024-06-21:
as for the vscode doesn't run in debian all you need to do just run it with `--no-sandbox` flag it sould work

closed 2024-06-23 by sabamdarif