sabamdarif/termux-desktop #8: Theme cairo not installed

binrusdi, 2024-06-19:
[image]

binrusdi, 2024-06-19:
i use MacOS Inspired-1 Style

sabamdarif, 2024-06-19:
run using these 

am start --user 0 -n com.termux.x11/com.termux.x11.MainActivity > /dev/null 2>&1
sleep 1
env DISPLAY=:1.0 dbus-launch --exit-with-session xfce4-session & > /dev/null 2>&1

sleep 5
process_id=$(ps -aux | grep '[x]fce4-screensaver' | awk '{print \$2}')
kill "$process_id" > /dev/null 2>&1

save it into a file and give it executable permission then run the file

sabamdarif, 2024-06-19:
> `WARNING: Some incorrect rendering might occur because the selected Vulkan device (Adreno (TM) 509) doesn't support base Zink requirements: feats.features.logicOp have_EXT_custom_border_color have_EXT_line_rasterization WARNING: Some incorrect rendering might occur because the selected Vulkan device (Adreno (TM) 509) doesn't support base Zink requirements: feats.features.logicOp have_EXT_custom_border_color have_EXT_line_rasterization ZINK: warning, this is cpu-based conditional rendering, say bye-bye to fps scalarBlockLayout OR EXT_scalar_block_layout warning : (/home/builder/.termux-build/cairo-dock-core/src/src/implementations/cairo-dock-egl.c:void gldi_register_egl_backend(void):232) Cairo-Dock was not built with EGL support WARNING: Some incorrect rendering might occur because the selected Vulkan device (Adreno (TM) 509) doesn't support base Zink requirements: feats.features.logicOp have_EXT_custom_border_color have_EXT_line_rasterization scalarBlockLayout OR EXT_scalar_block_layout warning : (/home/builder/.termux-build/cairo-dock-core/src/src/implementations/cairo-dock-glx.c:gboolean _initialize_opengl_backend(gboolean):129) couldn't find an appropriate visual, trying to get one without Stencil buffer (it may cause some little deterioration in the rendering) ...
> 
> # ============================================================================
> Cairo-Dock version : 3.4.99.alpha1
> Compiled date : May 13 2024 23:31:33
> Built with GTK : 3.24
> Running with OpenGL: 0
> g_file_test: assertion 'filename != NULL' failed g_file_test: assertion 'filename != NULL' failed`

why it starting the gpu server ?
how you launch the gui ?

i think you didn't understand 
Do this 

`cd $HOME`
`nano start`

paste this

```
am start --user 0 -n com.termux.x11/com.termux.x11.MainActivity > /dev/null 2>&1
sleep 1
env DISPLAY=:1.0 dbus-launch --exit-with-session xfce4-session & > /dev/null 2>&1

sleep 5
process_id=$(ps -aux | grep '[x]fce4-screensaver' | awk '{print $2}')
kill "$process_id" > /dev/null 2>&1
```

`chmod +x start`
`./start`

sabamdarif, 2024-07-06:
use the latest script and install it again 
if again it don't launch it with `--nogpu ` flag

binrusdi, 2024-07-15:
> use the latest script and install it again if again it didn't launch so launch it with `--nogpu ` flag

Do you have the latest script, but without reinstalling it?

sabamdarif, 2024-07-15:
Send me the current script you using to launch the termux x11 I will send you the new version of it 
Go to $PREFIX/bin
Type nano tx11start 
And send me what inside it

binrusdi, 2024-07-15:
sorry, It was deleted a long time ago... the file contents are empty :)

binrusdi, 2024-07-16:
What is the difference between zinc and virgl?

sabamdarif, 2024-07-17:
> What is the difference between zinc and virgl?

You can say it different methode to get hardware acceleration working in termux

closed 2024-07-23 by sabamdarif