sabamdarif/termux-desktop #13: gzip error

NoReplyUI5, 2024-07-07:
Install Gui Desktop In Termux                                          code by @sabamdrif

Downloading icon.tar.gz...                                             ################################################################ 100.0%                                                                       [80.6KiB/s] [===================================================>] 100%                                                                       gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now                             ~ $

sabamdarif, 2024-07-07:
which style ?

NoReplyUI5, 2024-07-07:
> which style ?

XFCE
style number 2: Modern Style

sabamdarif, 2024-07-07:
Fixed https://github.com/sabamdarif/termux-desktop/commit/94595ee706d20e67b2a012bae3af47f182f2f086

closed 2024-07-07 by sabamdarif