sabamdarif/termux-desktop #14: After Installation None of icon show on desktop with folder

Secondyou, 2024-07-19:
(it's happened randomly)

sabamdarif, 2024-07-19:
yes , it's a problem which i can find any fix yet , the only solution i found is to install a new icon theme

Secondyou, 2024-07-19:
i know a solution btw i think try to make script install fallback into papirus icon theme instead? if it's failed to extract and install

Secondyou, 2024-07-19:
(it's installable on pkg,apt or nala command)

sabamdarif, 2024-07-19:
ok 

> i know a solution btw i think try to make script install fallback into papirus icon theme instead? if it's failed to extract and install

yes it will be a good idia , but the icons croupted not because of it failed to install or something

sabamdarif, 2024-07-19:
also i onetime fix it another which work for me 
after a fresh install go to the icons folder and generate icon cache manually for each icon theme then start using the desktop

The main problem is something to do with the icon cache

Secondyou, 2024-07-20:
okay another solution is install fallback icon theme then gen cache after it installed
i thikk it's great

sabamdarif, 2024-07-20:
did you tried it ?

sabamdarif, 2024-07-20:
currently i didn't have this problem so i can't tell if it will work or not

closed 2024-07-20 by Secondyou