sabamdarif/termux-desktop #17: CANNOT LINK EXECUTABLE

Tausif-Husine, 2024-08-16:
After typing tx11start in the termux terminal I'm getting this error

CANNOT LINK EXECUTABLE "/data/data/com.termux/files/usr/bin/bash": library "/system/lib64/libskcodec.so" not found: needed by main executable

I've reinstalled it multiple times but it's not working and all the other commands for starting the gui isn't working either

sabamdarif, 2024-08-16:
which android device and termux version ?
OneUI i think

Tausif-Husine, 2024-08-16:
Samsung Galaxy A02 and termux version is 0.118 and yes oneui

sabamdarif, 2024-08-16:
which shell you are using bash or zsh ?
for zsh open `$HOME/.zshrc` and for bash open `/data/data/com.termux/files/usr/etc/bash.bashrc`
then there you find one line called `LD_PRELOAD=/system/lib64/libskcodec.so` comment or delete it.
then restart termux and launch the termux x11 again

Tausif-Husine, 2024-08-17:
nano $HOME/.zshrc
CANNOT LINK EXECUTABLE "nano": library "/system/lib64/libskcodec.so" not found: needed by main executable 🙂

sabamdarif, 2024-08-17:
Check if you can use sed command

Tausif-Husine, 2024-08-17:
I did it with fx explorer but now when I use tx11start it's giving me this error

"> "Warning:          Could not resolve keysym XF86NavInfo
"Errors from xkbcomp are not fatal to the X server"
vtest_client_dispatch_commands: client context created.
client: VTEST_CLIENT_DISCONNECTED
/data/data/com.termux/files/usr/bin/tx11start: line 27: 24506 Aborted                 env DISPLAY=:1.0 GALLIUM_DRIVER=virpipe MESA_GL_VERSION_OVERRIDE=4.0 dbus-launch --exit-with-session xfce4-session > /dev/null 2>&1

sabamdarif, 2024-08-17:
try running `tx11start --nogpu`

Tausif-Husine, 2024-08-17:
It's working but can't I do it with hardware acceleration?

sabamdarif, 2024-08-17:
which driver you select to get hardware acceleration

Tausif-Husine, 2024-08-17:
Vigril

sabamdarif, 2024-08-17:
try changeing it to zink

closed 2024-08-17 by Tausif-Husine