sabamdarif/termux-desktop #21: only termux cli look script.

raaz2507, 2024-09-04:
please add a script only for chage termux cli envirement . without intalling any desktop envirement , gpu exletion ect. 
only script for install zsh and other cli cusmization 
[image]

sabamdarif, 2024-09-04:
you can do it manually
 
1st) for zsh run this :- `pkg install zsh git -y`
then
`wget https://raw.githubusercontent.com/sabamdarif/simple-linux-scripts/main/install-zsh.sh && bash install-zsh.sh` 

2nd ) for color :- backup old color scheme by 
`mv $HOME/.termux/colors.properties $HOME/.termux/colors.properties.bak`
then run
`wget -O $HOME/.termux/colors.properties https://raw.githubusercontent.com/sabamdarif/termux-desktop/main/other/colors.properties`

3rd) then back old motd by 
`mv $PREFIX/etc/motd $PREFIX/etc/motd.bak ; mv $PREFIX/etc/motd-playstore $PREFIX/etc/motd-playstore.bak ; mv $PREFIX/etc/motd.sh $PREFIX/etc/motd.sh.bak` 
then
`wget -O $PREFIX/etc/motd.sh https://raw.githubusercontent.com/sabamdarif/termux-desktop/main/other/motd.sh`
and
`echo "bash $PREFIX/etc/motd.sh" >> "$PREFIX/etc/termux-login.sh"`

now restart termux and it's done

sabamdarif, 2024-09-16:
Script you requested :-  `wget https://raw.githubusercontent.com/sabamdarif/simple-linux-scripts/main/better-termux.sh && bash better-termux.sh`

closed 2024-09-16 by sabamdarif

raaz2507, 2024-09-28:
Thnx for your efforts.