sabamdarif/termux-desktop #23: error while configure python3-minimal

NickoDPu, 2024-09-08:
root@localhost:~# apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 python3 : PreDepends: python3-minimal (= 3.10.6-1~22.04) but 3.10.6-1~22.04.1 is installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
[image]

NickoDPu, 2024-09-08:
i want to install xfce4 and tigervnc, then i input command "apt update -y && apt upgrade -y" then the error come out. please some one can fix it

sabamdarif, 2024-09-08:
apt clean
apt --fix-broken install
dpkg --configure -a
sudo dpkg --force-all -i python3

NickoDPu, 2024-09-08:
ok work 😁

closed 2024-09-08 by sabamdarif