sabamdarif/termux-desktop #27: /data/data/com.termux/files/usr/bin/vncstart: line 13: syntax error near unexpected token `&' /data/data/com.termux/files/usr/bin/vncstart: line 13: `  &'

Wendel-JF, 2024-09-26:
This error is occurring when starting the server in vnc.
Here are some suggestions to fix this error:

1. in the termux-desktop terminal, enter this command.
`nano /data/data/com.termux/files/usr/bin/vncstart`

2. after put these codes:

`#!/data/data/com.termux/files/usr/bin/bash

export DISPLAY=:1

selected_hw_method="MESA_LOADER_DRIVER_OVERRIDE=zink"

case $1 in
  --nogpu)
    vncserver
    ;;
  --help|-h)
    echo "^[[1;32mType ^[[1;36mvncstart ^[[1;32mto start vncserver with gpu acceleration"^[[0m
    echo "^[[1;32mType ^[[1;36mvncstart --nogpu ^[[1;32mto start vncserver without gpu acceleration"^[[0m
    ;;
  *)
    case $DISPLAY in
      :*)
        MESA_LOADER_DRIVER_OVERRIDE=zink vncserver
        ;;
      *)
        echo "Display not supported"
        ;;
    esac
    ;;
esac`

3.press ctrl + o, enter and ctrl + x.

4.start the server with `vncstart` and see if it worked.

sabamdarif, 2024-09-28:
i will try

sabamdarif, 2024-10-01:
you get the syntax error because the hardware acceleration setup file didn't run perfectly.

there will be a line like 
MESA_NO_ERROR=1 MESA_GL_VERSION_OVERRIDE=4.3COMPAT MESA_GLES_VERSION_OVERRIDE=3.2 GALLIUM_DRIVER=zink ZINK_DESCRIPTORS=lazy virgl_test_server --use-egl-surfaceless --use-gles &

before MESA_LOADER_DRIVER_OVERRIDE=zink vncserver line

xgdgsc, 2024-10-17:
I also see this. And I feel the gpu acceleration step breaks the flow of the installer. And the instruction is a bit confusing for a new user. For example I still don' t know which to choose for my 8s gen 3 pad after reading.

sabamdarif, 2024-10-17:
> I also see this. And I feel the gpu acceleration step breaks the flow of the installer. And the instruction is a bit confusing for a new user. For example I still don' t know which to choose for my 8s gen 3 pad after reading.

which gpu do you have ?

xgdgsc, 2024-10-18:
Adreno 735 according to https://nanoreview.net/en/soc/qualcomm-snapdragon-8s-gen-3

sabamdarif, 2024-10-18:
> Adreno 735 according to https://nanoreview.net/en/soc/qualcomm-snapdragon-8s-gen-3

zink or virgl will work fine.

this mean you didn't the readme i added , i already show how to check your device gpu and also a performance test with different gpu drivers, 

follow this :- [here](https://github.com/sabamdarif/termux-desktop?tab=readme-ov-file#warning-follow-this-steps-point_down
)

xgdgsc, 2024-10-18:
Thanks. Works for me now.

closed 2024-10-18 by sabamdarif