sabamdarif/termux-desktop #29: Screen black

solakcemalettin, 2024-10-07:
[image]

sabamdarif, 2024-10-07:
`tx11start --nogpu` try this

solakcemalettin, 2024-10-07:
Same and showing this error [image]

sabamdarif, 2024-10-08:
Type `tdconfig` in termux and send me the part called device information

solakcemalettin, 2024-10-08:
[image]

[image]

sabamdarif, 2024-10-08:
i don't know  it might be the problem of your cpu architecture

daulatrathod, 2024-10-08:
Black screen 
[image]

daulatrathod, 2024-10-08:
tdconfig
[image]

daulatrathod, 2024-10-08:
Pls gave help me same on vnc

sfantu, 2024-10-08:
Try legacy drawing

ginestopo, 2024-10-19:
I am facing the same issue...

sabamdarif, 2024-10-20:
change ` XDG_RUNTIME_DIR=${TMPDIR} termux-x11 :1.0 &` with `XDG_RUNTIME_DIR=${TMPDIR} termux-x11 :1.0 -legacy-drawing &` in `$PREFIX/bin/tx11start` file

ginestopo, 2024-10-20:
@sabamdarif thanks a lot! Another problem I am facing is:
**[Process completed (signal 9) - press Enter]**
It shuts down termux (and termux:x11) randomly due to a phantom killer present in Android 12.

Although this error is not related to your script. I failed to find a safe and easy fix. (Access to developer setting is not possible in my device, cause I am running it in Meta Quest 3). This makes termux unusable in Android 12.

sabamdarif, 2024-10-20:
> @sabamdarif thanks a lot! Another problem I am facing is: **[Process completed (signal 9) - press Enter]** It shuts down termux (and termux:x11) randomly due to a phantom killer present in Android [...]

the method i knew that required usb debugging or wireless debugging or root to work

sabamdarif, 2024-10-20:
fixed https://github.com/sabamdarif/termux-desktop/commit/9dd423601fec5ba1165e362d10d9b63a1b357899

closed 2024-10-20 by sabamdarif