sabamdarif/termux-desktop #31: 安装失败

pengyixq, 2024-10-18:
安装过程中，有些链接无法获取，装上去后有些程序无法启动，能不能更新下安装代码呢

sabamdarif, 2024-10-19:
which links and programs can you , can you describe it properly .

closed 2024-11-16 by sabamdarif

pengyixq, 2024-11-24:
Thank you for your update and improvement. The previous issue has been resolved, but now I have found another problem. The software installed in Debian cannot be started on the Termux desktop without any response. I don't know the reason.

 
Closed #31 as completed.

sabamdarif, 2024-11-24:
> Thank you for your update and improvement. The previous issue has been resolved, but now I have found another problem. The software installed in Debian cannot be started on the Termux desktop without any response. I don't know the reason.
> […](#)

join the telegram channel