sabamdarif/termux-desktop #32: Black screeb abd EADDRINUSE error

ginestopo, 2024-10-19:
I get this error when launching termux. I configured it so it would automatically launch the x11

_XSERVTransSocketUNIXCreateListener: ...SocketCreateListener() failed
java.net.BindException: bind failed: EADDRINUSE (Address already in use)
        at libcore.io.IoBridge.bind(IoBridge.java:150)
        at java.net.PlainSocketImpl.socketBind(PlainSocketImpl.java:162)
        at java.net.AbstractPlainSocketImpl.bind(AbstractPlainSocketImpl.java:427)
        at java.net.ServerSocket.bind(ServerSocket.java:377)
        at java.net.ServerSocket.<init>(ServerSocket.java:237)
        at com.termux.x11.CmdEntryPoint.lambda$spawnListeningThread$1(CmdEntryPoint.java:144)
        at com.termux.x11.CmdEntryPoint.$r8$lambda$ViPcpFnEzoJCKyl-YNiduQ0Vo1o(CmdEntryPoint.java:0)
        at com.termux.x11.CmdEntryPoint$$ExternalSyntheticLambda2.run(R8$$SyntheticClass:0)
        at java.lang.Thread.run(Thread.java:920)
Caused by: android.system.ErrnoException: bind failed: EADDRINUSE (Address already in use)
        at libcore.io.Linux.bind(Native Method)
        at libcore.io.ForwardingOs.bind(ForwardingOs.java:141)
        at libcore.io.IoBridge.bind(IoBridge.java:146)
        ... 8 more

sabamdarif, 2024-10-20:
`tx11stop -f` run this and try again

sabamdarif, 2024-10-20:
fixed https://github.com/sabamdarif/termux-desktop/commit/9dd423601fec5ba1165e362d10d9b63a1b357899