sabamdarif/termux-desktop #34: cannot use backspace in X

darvari, 2024-10-30:
tx11start in xfce

example1
open terminal in gui
type something
try to delete a few characters <--no reaction 

example2
open file-manager
click in address bar
try to delete something <-- does not work

sabamdarif, 2024-10-30:
Open termux x11 preferences -> keyboard -> workaround to enable cjk gboard ( enable it)

darvari, 2024-10-30:
Thanks for the quick answer.
I tried exactly that, but it didn't work.
It says "may require Android 14 and Gboard14"
But I am on LineageOS 18, which is Android 11 and I use Openboard as keyboard - any chance for me?

sabamdarif, 2024-10-31:
> Thanks for the quick answer. I tried exactly that, but it didn't work. It says "may require Android 14 and Gboard14" But I am on LineageOS 18, which is Android 11 and I use Openboard as keyboard - any chance for me?

use [FlorisBoard](https://github.com/florisboard/florisboard) i am also using it with a custom rom or use [hackers keyboaed](https://store.nethunter.com/packages/org.pocketworkstation.pckeyboard/) (best with termux)

darvari, 2024-10-31:
I tried FlorisBoard and it seems to work fine enough for me.
Thank you!

closed 2024-10-31 by sabamdarif