sabamdarif/termux-desktop #35: Termux error

astroskyoffical, 2024-11-06:
Samsung s24 snapdragon 8 gen 3

Termux  and x11 Only 

Used your script. Enabled freedreno and zink respectfully from the auto setup 

Disabled auto start for debugging purposes 

Tried running the manual start with GPU and with out

Regardless of that I get this error

[image]

And for my config file it shows this 

[image]

I understand that the ardeno 750 isnt as well supported as others but there's been quite a few changes made from others that have managed to get it to working very well.  

Any ideas why I'm facing this issue?

astroskyoffical, 2024-11-06:
Just for reference I manually ran the change hW command as I read the other closed support threads here and you mentioned that can fix things like this.  Unfortunately this did nothing for me and actually it said it removed alot of things? Not sure.   Definitely buggy

sabamdarif, 2024-11-06:
> Samsung s24 snapdragon 8 gen 3
> 
> Termux and x11 Only
> 
> Used your script. Enabled freedreno and zink respectfully from the auto setup
> 
> Disabled auto start for debugging purposes
> 
> Tried running the manual start with GPU and with out
> 
> Regardless of that I get this error
> 
> ![Screenshot_20241106_093230_Termux](https://private-user-images.githubusercontent.com/101475363/383622701-6ab965bf-0fbf-4322-8d6f-9a88194356b5.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA5MTIyMDcsIm5iZiI6MTczMDkxMTkwNywicGF0aCI6Ii8xMDE0NzUzNjMvMzgzNjIyNzAxLTZhYjk2NWJmLTBmYmYtNDMyMi04ZDZmLTlhODgxOTQzNTZiNS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwNlQxNjUxNDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lY2E3MzRkYjk1MTQ2MmI3ODBhYmZhOWRmZDNlYjQ5YWY3ZWVlMjI4YmM4MDBhYTdkZjEwYWJiNjIxZWQzM2VmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.rmt3Dbpc8xU8Btc9VAjollnBFIqTcWrKN2FWh1_I6cA)
> 
> And for my config file it shows this
> 
> ![Screenshot_20241106_093407_Termux](https://private-user-images.githubusercontent.com/101475363/383623251-2d77708d-1f99-4c31-93ad-bec26b8fe571.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA5MTIyMDcsIm5iZiI6MTczMDkxMTkwNywicGF0aCI6Ii8xMDE0NzUzNjMvMzgzNjIzMjUxLTJkNzc3MDhkLTFmOTktNGMzMS05M2FkLWJlYzI2YjhmZTU3MS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwNlQxNjUxNDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04YmMxMjAxMmU1YWEwODlhMmRjODExM2JjZjM0ZjE5YjJiMjA1MmFhZGZmNTZiMzM0MjcxM2UyMTU0NTE2ODE0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.g2LSHz67SmqjY8E2ZUg8eElNJ_USb2dnYv5zD8Jf2LA)
> 
> I understand that the ardeno 750 isnt as well supported as others but there's been quite a few changes made from others that have managed to get it to working very well.
> 
> Any ideas why I'm facing this issue?

sorry it's a mistake i made i have fix it in the screenshot you can see empty `&`  before each `XDG_RUNTIME_DIR` remove those `&` then save it and launch termux x11 again

astroskyoffical, 2024-11-06:
Thanks but unfortunately I don't have hardware acceleration it seems.   Despite installing zink .   It's defaulting to cpu ONLY.   What can I do? 
[image]

Also I can't install packages the way you mention in the guide in proot. I have to run sudo for some reason 
[image]

astroskyoffical, 2024-11-06:
Here's me trying to run supertuckart and this is the output in the terminal outside of the x11

### Issue Description

I encountered an issue while running `tx11start` with the following output:
~ $ Warning: CPU doesn't support 32-bit instructions, some software may not work.
~ $ tx11startproot warning: can't sanitize binding "/proc/self/fd/2": No such file or directory
proot warning: can't sanitize binding "/proc/self/fd/1": No such file or directory
[verbose  ] main: Error messages and other text output will be logged to /home/astro/.config/supertuxkart/config-0.10/stdout.log.
[info   ] [FileManager]: Data files will be fetched from: '/usr/share/games/supertuxkart/data/'
[info   ] [FileManager]: User directory is '/home/astro/.config/supertuxkart/config-0.10/'.
[info   ] [FileManager]: Addons files will be stored in '/home/astro/.local/share/supertuxkart/addons/'.
[info   ] [FileManager]: Screenshots will be stored in '/home/astro/.local/share/supertuxkart/screenshots/'.
[info   ] [FileManager]: User-defined grand prix will be stored in '/home/astro/.local/share/supertuxkart/grandprix/'.
[info   ] [FileManager]: Asset 0 will be loaded from '/usr/share/games/supertuxkart/data/challenges/'.
[info   ] [FileManager]: Asset 1 will be loaded from '/usr/share/games/supertuxkart/data/gfx/'.
[info   ] [FileManager]: Asset 2 will be loaded from '/usr/share/games/supertuxkart/data/grandprix/'.
[info   ] [FileManager]: Asset 3 will be loaded from '/usr/share/games/supertuxkart/data/gui/icons/'.
[info   ] [FileManager]: Asset 4 will be loaded from '/usr/share/games/supertuxkart/data/gui/screens/'.
[info   ] [FileManager]: Asset 5 will be loaded from '/usr/share/games/supertuxkart/data/gui/dialogs/'.
[info   ] [FileManager]: Asset 6 will be loaded from '/usr/share/games/supertuxkart/data/replay/'.
[info   ] [FileManager]: Asset 7 will be loaded from '/usr/share/games/supertuxkart/data/shaders/'.
[info   ] [FileManager]: Asset 8 will be loaded from '/usr/share/games/supertuxkart/data/skins/'.
[info   ] [FileManager]: Asset 9 will be loaded from '/usr/share/games/supertuxkart/data/ttf/'.
[info   ] [FileManager]: Asset 10 will be loaded from '/usr/share/games/supertuxkart/data/po/'.
[info   ] [FileManager]: Asset 11 will be loaded from '/usr/share/games/supertuxkart/data/library/'.
[info   ] [FileManager]: Asset 12 will be loaded from '/usr/share/games/supertuxkart/data/models/'.
[info   ] [FileManager]: Asset 13 will be loaded from '/usr/share/games/supertuxkart/data/music/'.
[info   ] [FileManager]: Asset 14 will be loaded from '/usr/share/games/supertuxkart/data/sfx/'.
[info   ] [FileManager]: Asset 15 will be loaded from '/usr/share/games/supertuxkart/data/textures/'.
[info   ] [FileManager]: Asset 16 will be loaded from '/usr/share/games/supertuxkart/data/tracks/'.
[verbose  ] translation: Env var LANGUAGE = 'en_US.UTF-8'.
[verbose  ] translation: Language 'English (United States)'.
..:: Antarctica Rendering Engine 2.0 ::..
[info   ] [IrrDriver Logger]: ..:: Antarctica Rendering Engine 2.0 ::..
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
libGL error: glx: failed to create drisw screen
libGL error: failed to load driver: swrast
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  150 (GLX)
  Minor opcode of failed request:  3 (X_GLXCreateContext)
  Value in failed request:  0x0
  Serial number of failed request:  116
  Current serial number in output stream:  117
~ $ tx11start

astroskyoffical, 2024-11-06:
I tried to install termux nodded mesa drivers for the 750. I forced installed it through proot.  Super tux kart launches oddly but it has no window it's just around and it acts super odd.   

Error is this now 

[image]

_bgra
[info   ] IrrDriver: OpenGL version: 3.2
[info   ] IrrDriver: OpenGL vendor: Collabora Ltd
[info   ] IrrDriver: OpenGL renderer: zink (Turnip Adreno (TM) 750)
[info   ] IrrDriver: OpenGL version string: OpenGL ES 3.2 Mesa 22.3.6
[info   ] GLDriver: Explicit Attrib Location Present
[info   ] GLDriver: ARB Uniform Buffer Object Present
[info   ] GLDriver: EXT texture format BGRA8888 Present
[info   ] GLDriver: EXT Color Buffer Float Present
[info   ] GLDriver: EXT Texture Compression S3TC Present
[info   ] GLDriver: EXT Texture Compression S3TC sRGB Present
[info   ] SharedGPUObjects: Hardware Skinning enabled, method: 1024 (max bones) * 16 RGBA float texture
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/screenquad.vert
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/motion_blur.frag
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/lightning.frag
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/primitive2dlist.vert
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/transparent.frag
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/texturedquad.vert
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/uniformcolortexturedquad.frag
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/texturedquad.frag
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/coloredquad.vert
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/coloredquad.frag
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/colortexturedquad.vert
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/colortexturedquad.frag
[info   ] irr_driver: GLSL supported.
[error  ] InputManager: Failed to init SDL haptics: Could not initialize UDEV
[info   ] GUI: Loading skin data from file: /usr/share/games/supertuxkart/data/skins/peach/stkskin.xml
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/sp_pass.vert
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/sp_solid.frag
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/sp_shadow.vert
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/white.frag
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/sp_skinning.vert
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/sp_skinning_shadow.vert
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/sp_normal_map.frag
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/sp_alpha_test.frag
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/sp_shadow_alpha_test.frag
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/sp_decal.frag
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/sp_grass_pass.vert
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/sp_grass.frag
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/sp_grass_shadow.vert
[info   ] ShaderFilesManager: Compiling shader: /usr/share/games/supertuxkart/data/shaders/sp_unlit.frag
~ $

Only enough supertuxkart in native launches but no hardware acceleration just cpu. So slow