sabamdarif/termux-desktop #36: Desktop Crashing

Lobo6780, 2024-11-06:
Two different phones:
- Motorola Edge 50 Pro 5G
- Redmi Note 8 Pro 6GB

Motorola installed with adreno drivers and on Redmi Virgl renderer, desktop runs only for about 2 minutes and after that X11 closes or VNC connection closes with this same errors on two devices:

`call functions->texImage2D(nativegl::GetTextureBindingTarget(target), static_cast<GLint>(level), texImageFormat.internalFormat, size.width, size.height, 0, texImageFormat.format, texImageFormat.type, pixels) generated error 0x00000502 in ../../../cache/tmp-checkout/angle/src/libANGLE/renderer/gl/TextureGL.cpp, setImageHelper:276.
ERR: renderergl_utils.cpp:3057 (HandleError): GL call functions->texImage2D(nativegl::GetTextureBindingTarget(target), static_cast<GLint>(level), texImageFormat.internalFormat, size.width, size.height, 0, texImageFormat.format, texImageFormat.type, pixels) generated error 0x00000502 in ../../../cache/tmp-checkout/angle/src/libANGLE/renderer/gl/TextureGL.cpp, setImageHelper:276.
ERR: renderergl_utils.cpp:3057 (HandleError): GL call functions->texImage2D(nativegl::GetTextureBindingTarget(target), static_cast<GLint>(level), texImageFormat.internalFormat, size.width, size.height, 0, texImageFormat.format, texImageFormat.type, pixels) generated error 0x00000502 in ../../../cache/tmp-checkout/angle/src/libANGLE/renderer/gl/TextureGL.cpp, setImageHelper:276.
ERR: renderergl_utils.cpp:3057 (HandleError): GL call functions->texImage2D(nativegl::GetTextureBindingTarget(target), static_cast<GLint>(level), texImageFormat.internalFormat, size.width, size.height, 0, texImageFormat.format, texImageFormat.type, pixels) generated error 0x00000502 in ../../../cache/tmp-checkout/angle/src/libANGLE/renderer/gl/TextureGL.cpp, setImageHelper:276.
ERR: renderergl_utils.cpp:3057 (HandleError): GL call functions->texImage2D(nativegl::GetTextureBindingTarget(target), static_cast<GLint>(level), texImageFormat.internalFormat, size.width, size.height, 0, texImageFormat.format, texImageFormat.type, pixels) generated error 0x00000502 in ../../../cache/tmp-checkout/angle/src/libANGLE/renderer/gl/TextureGL.cpp, setImageHelper:276.
ERR: renderergl_utils.cpp:3057 (HandleError): GL call functions->texImage2D(nativegl::GetTextureBindingTarget(target), static_cast<GLint>(level), texImageFormat.internalFormat, size.width, size.height, 0, texImageFormat.format, texImageFormat.type, pixels) generated error 0x00000502 in ../../../cache/tmp-checkout/angle/src/libANGLE/renderer/gl/TextureGL.cpp, setImageHelper:276.
vtest_client_dispatch_commands: client context created.
vrend_winsys_make_context_current: Error switching context: EGL_BAD_ACCESS
ERR: renderergl_utils.cpp:3057 (HandleError): GL call functions->texImage2D(nativegl::GetTextureBindingTarget(target), static_cast<GLint>(level), texImageFormat.internalFormat, size.width, size.height, 0, texImageFormat.format, texImageFormat.type, pixels) generated error 0x00000502 in ../../../cache/tmp-checkout/angle/src/libANGLE/renderer/gl/TextureGL.cpp, setImageHelper:276.
ERR: renderergl_utils.cpp:3057 (HandleError): GL call functions->texImage2D(nativegl::GetTextureBindingTarget(target), static_cast<GLint>(level), texImageFormat.internalFormat, size.width, size.height, 0, texImageFormat.format, texImageFormat.type, pixels) generated error 0x00000502 in ../../../cache/tmp-checkout/angle/src/libANGLE/renderer/gl/TextureGL.cpp, setImageHelper:276.
ERR: renderergl_utils.cpp:3057 (HandleError): GL call functions->texImage2D(nativegl::GetTextureBindingTarget(target), static_cast<GLint>(level), texImageFormat.internalFormat, size.width, size.height, 0, texImageFormat.format, texImageFormat.type, pixels) generated error 0x00000502 in ../../../cache/tmp-checkout/angle/src/libANGLE/renderer/gl/TextureGL.cpp, setImageHelper:276.
ERR: renderergl_utils.cpp:3057 (HandleError): GL call functions->texImage2D(nativegl::GetTextureBindingTarget(target), static_cast<GLint>(level), texImageFormat.internalFormat, size.width, size.height, 0, texImageFormat.format, texImageFormat.type, pixels) generated error 0x00000502 in ../../../cache/tmp-checkout/angle/src/libANGLE/renderer/gl/TextureGL.cpp, setImageHelper:276.
ERR: renderergl_utils.cpp:3057 (HandleError): GL call functions->texImage2D(nativegl::GetTextureBindingTarget(target), static_cast<GLint>(level), texImageFormat.internalFormat, size.width, size.height, 0, texImageFormat.format, texImageFormat.type, pixels) generated error 0x00000502 in ../../../cache/tmp-checkout/angle/src/libANGLE/renderer/gl/TextureGL.cpp, setImageHelper:276.
ERR: renderergl_utils.cpp:3057 (HandleError): GL call functions->texImage2D(nativegl::GetTextureBindingTarget(target), static_cast<GLint>(level), texImageFormat.internalFormat, size.width, size.height, 0, texImageFormat.format, texImageFormat.type, pixels) generated error 0x00000502 in ../../../cache/tmp-checkout/angle/src/libANGLE/renderer/gl/TextureGL.cpp, setImageHelper:276.
vtest_client_dispatch_commands: client context created.
vrend_winsys_make_context_current: Error switching context: EGL_BAD_ACCESS
ERR: renderergl_utils.cpp:3057 (HandleError): GL call functions->texImage2D(nativegl::GetTextureBindingTarget(target), static_cast<GLint>(level), texImageFormat.internalFormat, size.width, size.height, 0, texImageFormat.format, texImageFormat.type, pixels) generated error 0x00000502 in ../../../cache/tmp-checkout/angle/src/libANGLE/renderer/gl/TextureGL.cpp, setImageHelper:276.
ERR: renderergl_utils.cpp:3057 (HandleError): GL call functions->texImage2D(nativegl::GetTextureBindingTarget(target), static_cast<GLint>(level), texImageFormat.internalFormat, size.width, size.height, 0, texImageFormat.format, texImageFormat.type, pixels) generated error 0x00000502 in ../../../cache/tmp-checkout/angle/src/libANGLE/renderer/gl/TextureGL.cpp, setImageHelper:276.
ERR: renderergl_utils.cpp:3057 (HandleError): GL call functions->texImage2D(nativegl::GetTextureBindingTarget(target), static_cast<GLint>(level), texImageFormat.internalFormat, size.width, size.height, 0, texImageFormat.format, texImageFormat.type, pixels) generated error 0x00000502 in ../../../cache/tmp-checkout/angle/src/libANGLE/renderer/gl/TextureGL.cpp, setImageHelper:276.
ERR: renderergl_utils.cpp:3057 (HandleError): GL call functions->texImage2D(nativegl::GetTextureBindingTarget(target), static_cast<GLint>(level), texImageFormat.internalFormat, size.width, size.height, 0, texImageFormat.format, texImageFormat.type, pixels) generated error 0x00000502 in ../../../cache/tmp-checkout/angle/src/libANGLE/renderer/gl/TextureGL.cpp, setImageHelper:276.
ERR: renderergl_utils.cpp:3057 (HandleError): GL call functions->texImage2D(nativegl::GetTextureBindingTarget(target), static_cast<GLint>(level), texImageFormat.internalFormat, size.width, size.height, 0, texImageFormat.format, texImageFormat.type, pixels) generated error 0x00000502 in ../../../cache/tmp-checkout/angle/src/libANGLE/renderer/gl/TextureGL.cpp, setImageHelper:276.
ERR: renderergl_utils.cpp:3057 (HandleError): GL call functions->texImage2D(nativegl::GetTextureBindingTarget(target), static_cast<GLint>(level), texImageFormat.internalFormat, size.width, size.height, 0, texImageFormat.format, texImageFormat.type, pixels) generated error 0x00000502 in ../../../cache/tmp-checkout/angle/src/libANGLE/renderer/gl/TextureGL.cpp, setImageHelper:276.
vtest_client_dispatch_commands: client context created.
vrend_winsys_make_context_current: Error switching context: EGL_BAD_ACCESS
client: VTEST_CLIENT_DISCONNECTED`

sabamdarif, 2024-11-06:
try running them without gpu acceleration using `tx11start --nogpu` or `vncstart --nogpu`

Lobo6780, 2024-11-06:
Tried it, crashed again, this time no error only Process completed (signal 9)

sabamdarif, 2024-11-06:
> Tried it, crashed again, this time no error only Process completed (signal 9)

disable Phantom Processes Killer using adb
https://github.com/atamshkai/Phantom-Process-Killer

Lobo6780, 2024-11-06:
Thank you I think worked! No crash for 10 minutes.

Lobo6780, 2024-11-06:
> > Tried it, crashed again, this time no error only Process completed (signal 9)
> 
> disable Phantom Processes Killer using adb https://github.com/atamshkai/Phantom-Process-Killer

closed 2024-11-06 by Lobo6780