sabamdarif/termux-desktop #37: What is the current state of s24 support

astroskyoffical, 2024-11-07:
My last open request was closed but is there a way to get the 750 to be supported in native mode?

sabamdarif, 2024-11-07:
use zink or virgl it will be best

astroskyoffical, 2024-11-07:
> use zink or virgl it will be best

Thanks however despite selecting zink and then forcing zink through the change hardware script you provided. I'm still only getting this
[image]

sabamdarif, 2024-11-07:
open tx11start file and send what you have inside that

sabamdarif, 2024-11-07:
also for freedreno i don't work in most of the phones, i have to find a solution for that it need some time

astroskyoffical, 2024-11-07:
#!/data/data/com.termux/files/usr/bin/bash

case $1 in
--nogpu)
XDG_RUNTIME_DIR=${TMPDIR} termux-x11 :1.0 &
sleep 1
am start --user 0 -n com.termux.x11/com.termux.x11.MainActivity > /dev/null 2>&1
sleep 1
env DISPLAY=:1.0 dbus-launch --exit-with-session xfce4-session > /dev/null 2>&1 &
    if [[ "$2" == "--legacy" ]]; then
        XDG_RUNTIME_DIR=${TMPDIR} termux-x11 :1.0 -legacy-drawing &
        sleep 1
        am start --user 0 -n com.termux.x11/com.termux.x11.MainActivity > /dev/null 2>&1
        sleep 1
        env DISPLAY=:1.0 dbus-launch --exit-with-session xfce4-session > /dev/null 2>&1 &
    fi
;;
--nodbus)
  XDG_RUNTIME_DIR=${TMPDIR} GALLIUM_DRIVER=zink MESA_GL_VERSION_OVERRIDE=4.0 termux-x11 :1 -xstartup "xfce4-session"
exit 0
    case $2 in
        --nogpu)
        XDG_RUNTIME_DIR=${TMPDIR} termux-x11 :1 -xstartup "xfce4-session"
            if [[ "$3" == "--legacy" ]]; then
                XDG_RUNTIME_DIR=${TMPDIR} termux-x11 :1 -legacy-drawing -xstartup "xfce4-session"
            fi
        ;;
        --legacy)
          XDG_RUNTIME_DIR=${TMPDIR} GALLIUM_DRIVER=zink MESA_GL_VERSION_OVERRIDE=4.0 termux-x11 :1 -legacy-drawing -xstartup "xfce4-session"
        ;;
        *)
        echo -e "\033[1;36m--legacy \033[1;32mto start termux:x11 with -legacy-drawing\033[0m"
        echo -e "\033[1;36m---nogpu \033[1;32mto start termux:x11 without gpu acceleration\033[0m"
        exit 0
        ;;
    esac
;;
--legacy)
  XDG_RUNTIME_DIR=${TMPDIR} termux-x11 :1 -legacy-drawing &
sleep 1
am start --user 0 -n com.termux.x11/com.termux.x11.MainActivity > /dev/null 2>&1 &
sleep 1
env DISPLAY=:1.0 MESA_LOADER_DRIVER_OVERRIDE=zink dbus-launch --exit-with-session xfce4-session > /dev/null 2>&1 &
;;
--help|-h)
echo -e "\033[1;36mtx11start \033[1;32mto start termux:x11 with gpu acceleration\033[0m"
echo -e "\033[1;36mtx11start --nogpu \033[1;32mto start termux:x11 without gpu acceleration\033[0m"
echo -e "\033[1;36mtx11start --nodbus \033[1;32mto start termux:x11 without gpu dbus\033[0m"
echo -e "\033[1;36mtx11start --legacy \033[1;32mto start termux:x11 with -legacy-drawing\033[0m"
exit 0
;;
*)
  XDG_RUNTIME_DIR=${TMPDIR} termux-x11 :1 &
sleep 1
am start --user 0 -n com.termux.x11/com.termux.x11.MainActivity > /dev/null 2>&1 &
sleep 1
env DISPLAY=:1.0 MESA_LOADER_DRIVER_OVERRIDE=zink dbus-launch --exit-with-session xfce4-session > /dev/null 2>&1 &
;;
esac
sleep 5
process_id=$(ps -aux | grep '[x]fce4-screensaver' | awk '{print $2}')
kill "$process_id" > /dev/null 2>&1

astroskyoffical, 2024-11-07:
Thanks for trying so hard man. Your script has been the easiest one online and seems thanks complete.  I try to make this a real computer with Samsung dex and having full hardware acceleration would be amazing! I like to try to run games and port over xemu over to the phone through termux.

astroskyoffical, 2024-11-07:
> also for freedreno it don't work in most of the phones, i have to find a solution for that it need some time

Sent above

sabamdarif, 2024-11-07:
run `setup-termux-desktop -c hw` then type y then for select zink or virgl and let it do it's thing after that try agin

sabamdarif, 2024-11-07:
> Thanks for trying so hard man. Your script has been the easiest one online and seems thanks complete. I try to make this a real computer with Samsung dex and having full hardware acceleration would be amazing! I like to try to run games and port over xemu over to the phone through termux.

thanks

astroskyoffical, 2024-11-08:
Ran the command. Selected zink. It said it was installing alot of software that didn't seem like it should be removed.  Now when I click start I get a blank screen and cursor. Tried the stop command and forcing stop termux. No good .....

astroskyoffical, 2024-11-08:
Ok good news I tried pdrun supertuxkart and randomly it started showing the entire game In x11 and it seems like its using my hardware as it's smoother.  Unfortunately idk how to get the desktop back

astroskyoffical, 2024-11-08:
> > Thanks for trying so hard man. Your script has been the easiest one online and seems thanks complete. I try to make this a real computer with Samsung dex and having full hardware acceleration would be amazing! I like to try to run games and port over xemu over to the phone through termux.
> 
> thanks

Ok new update.    Tx11start --nogpu works

This is great but not the intended behavior I want....I want hardware acceleration. I selected zink......so what gives?
[image]

astroskyoffical, 2024-11-08:
Ok it now outputs an error when I run the normal start command

~ $ tx11start
ZINK: USING LAZY DESCRIPTORS
../src/src/gallium/drivers/zink/zink_resource.c:804: struct zink_resource_object *resource_object_create(struct zink_screen *, const struct pipe_resource *, struct winsys_handle *, _Bool *, const uint64_t *, int): assertion "reqs.memoryTypeBits & BITFIELD_BIT(mai.memoryTypeIndex)" failed
/data/data/com.termux/files/usr/bin/tx11start: line 66: 10255 Aborted                 env DISPLAY=:1.0 GALLIUM_DRIVER=virpipe MESA_GL_VERSION_OVERRIDE=4.0 dbus-launch --exit-with-session xfce4-session > /dev/null 2>&1
~ $

sabamdarif, 2024-11-08:
in termux run `pkg install xfce4 xfce4-goodies -y`

astroskyoffical, 2024-11-08:
> in termux run `pkg install xfce4 xfce4-goodies -y`

Hey thanks for the reply. I did some research and the freedrino package you install through your script  is likely the reason it doesn't work on 750.   Not only that but the main package provided by termux has yet to be updated to the latest main mesa branch including the relevant patches so unfortunately it has to be manually compiled and installed.  I believe it starts from mesa 24.02 and above for full 750 support.  Perhaps you can assist or I can help you compile it and allow it as an option in your script?

sabamdarif, 2024-11-08:
https://t.me/saba_mdarif come here

sabamdarif, 2024-11-09:
add turnip setup script https://github.com/sabamdarif/termux-desktop/commit/261ef3be0ff0004c7f7ac6408cc186adfe153459 now it will be much better

closed 2024-11-09 by sabamdarif