sabamdarif/termux-desktop #39: Error when i type wine winecfg

Yamak5, 2024-11-10:
How can i run exe ?
It gave me error when i type wine winecfg 
Like that :
wine winecfg
/data/data/com.termux/files/usr/bin/wine: line 52: /data/data/com.termux/files/usr/glibc/opt/package-manager/token: No such file or directory
/data/data/com.termux/files/usr/bin/wine: line 81: /data/data/com.termux/files/usr/glibc/opt/scripts/configs: No such file or directory
/data/data/com.termux/files/usr/bin/wine: line 82: load_configs: command not found
chmod: cannot access '/data/data/com.termux/files/usr/glibc/bin/box64': No such file or directory
chmod: cannot access '/bin/wine': No such file or directory
chmod: cannot access '/bin/wineserver': No such file or directory
/data/data/com.termux/files/usr/bin/wine: line 97: patchelf: command not found
ln: failed to create symbolic link '/data/data/com.termux/files/usr/glibc/bin/wine': No such file or directory
ln: failed to create symbolic link '/data/data/com.termux/files/usr/glibc/bin/wineserver': No such file or directory
cp: cannot stat '/data/data/com.termux/files/usr/glibc/opt/prefix/start/Registry/2.*': No such file or directory
cp: cannot stat '/data/data/com.termux/files/usr/glibc/opt/prefix/start/Install/media foundation (for RE).lnk': No such file or directory
mkdir: cannot create directory ‘/moboxmeta’: Read-only file system
Applying services fix
touch: cannot touch '/moboxmeta/services-fix-applied': No such file or directory
Applying fonts fix
tar: /data/data/com.termux/files/usr/glibc/opt/prefix/fix-fonts.tar.xz: Cannot open: No such file or directory
tar: Error is not recoverable: exiting now
touch: cannot touch '/moboxmeta/fonts-fix-applied': No such file or directory
Applying dlls fix
touch: cannot touch '/moboxmeta/dxdlls-fix-applied': No such file or directory
/data/data/com.termux/files/usr/bin/wine: line 157: load_configs: command not found
taskset: failed to parse CPU list: /data/data/com.termux/files/usr/glibc/bin/box64

How can i fix it :(
[image]

sabamdarif, 2024-11-10:
try reinstalling mobox, after reinstalling mobox launch it once using mobox command

Yamak5, 2024-11-10:
When i type mobox in terminal that's happen :
[image]

And i wait a little the this happen :

[image]

sabamdarif, 2024-11-10:
bro type mobox command in termux terminal (no need to do it now , what you did is also fine) not in the desktop now stop termux x11(type tx11stop -f  in termux) then start termux desktop and run winecfg

Yamak5, 2024-11-10:
Everything is fine now !
Thank You So much :)

closed 2024-11-10 by sabamdarif

Yamak5, 2024-11-10:
Does Virgl zink support Mali GPU or only Adreno
And what about Virgl does support Mali?

And the VirGL Render run in apps and firefox and games automatically when i open them or i need to type some codes to run them in VirGL or they will run in llvmpipe automatically ?

gurachan, 2025-02-08:
> bro type mobox command in termux terminal (no need to do it now , what you did is also fine) not in the desktop now stop termux x11(type tx11stop -f in termux) then start termux desktop and run winecfg

I did that but still no wine..

sabamdarif, 2025-02-08:
> > bro type mobox command in termux terminal (no need to do it now , what you did is also fine) not in the desktop now stop termux x11(type tx11stop -f in termux) then start termux desktop and run winecfg
> 
> I did that but still no wine..

with mobox ?
did you start mobox once before starting it in the desktop ?