sabamdarif/termux-desktop #40: Error Render in VirGL Zink for (Mali-g52)

Yamak5, 2024-11-11:
Does Virgl zink support Mali-g52 GPU or only Adreno
And what about Virgl does support Mali?

And the VirGL Render run in apps and firefox and games automatically when i open them or i need to type some codes to run them in VirGL or they will run in llvmpipe automatically ?

I got this evey time i open Termux gui 

[image]

sabamdarif, 2024-11-11:
it have no fix for now just wait until termux team fix it

Yamak5, 2024-11-11:
So I understand from you that there is no freegel supporting my money

Yamak5, 2024-11-11:
So, I understand from you that there are no Mesa drivers, like VirGL, that support the Mali GPU."

sabamdarif, 2024-11-11:
> So, I understand from you that there are no Mesa drivers, like VirGL, that support the Mali GPU."

Some mali gpu suppot some not

closed 2024-11-16 by sabamdarif