sabamdarif/termux-desktop #41: Can't run pdrun

RimaStudioo, 2024-11-11:
I have problems running pdrun and ubuntu commands, devices huawei matepad 11, snapdragon 860, andreno 650, I use graphic turnip and freedreno

[image]

[image]

[image]

sabamdarif, 2024-11-11:
go to $PREFIX/bin and send me what inside pdrun file

RimaStudioo, 2024-11-11:
#!/data/data/com.termux/files/usr/bin/bash

selected_pd_hw_method="MESA_LOADER_DRIVER_OVERRIDE=zink TU_DEBUG=noconform"
bash ~/.pd-sound-service
xhost + > /dev/null 2>&1
case $1 in
--nogpu)
proot-distro login --user  ubuntu --shared-tmp -- env DISPLAY=:1.0 ${@:2}
;;
--help|-h)
echo "[1;32mType [1;36mpdrun launch command [1;32mto start with gpu acceleration"[0m
echo "[1;32mType [1;36mpdrun ---nogpu launch command [1;32mto start without gpu acceleration"[0m
;;
*)
proot-distro login --user  ubuntu --shared-tmp -- eval "env DISPLAY=:1.0 $selected_pd_hw_method $@"
;;
esac

RimaStudioo, 2024-11-11:
> go to $PREFIX/bin and send me what inside pdrun file

#!/data/data/com.termux/files/usr/bin/bash

selected_pd_hw_method="MESA_LOADER_DRIVER_OVERRIDE=zink TU_DEBUG=noconform"
bash ~/.pd-sound-service
xhost + > /dev/null 2>&1
case $1 in
--nogpu)
proot-distro login --user  ubuntu --shared-tmp -- env DISPLAY=:1.0 ${@:2}
;;
--help|-h)
echo "[1;32mType [1;36mpdrun launch command [1;32mto start with gpu acceleration"[0m
echo "[1;32mType [1;36mpdrun ---nogpu launch command [1;32mto start without gpu acceleration"[0m
;;
*)
proot-distro login --user  ubuntu --shared-tmp -- eval "env DISPLAY=:1.0 $selected_pd_hw_method $@"
;;
esac

RimaStudioo, 2024-11-11:
**inside ubuntu file**

#!/data/data/com.termux/files/usr/bin/bash

if [[ "$#" -eq 0 ]]; then
    proot-distro login --user  ubuntu --shared-tmp
elif [[ "$1" = "install" ]]; then
if [[ ! -d "/data/data/com.termux/files/usr/share/applications/pd_added" ]]; then
        mkdir -p "/data/data/com.termux/files/usr/share/applications/pd_added"
        fi
proot-distro login --user  ubuntu --shared-tmp -- env DISPLAY=:1.0 sudo nala install ${@:2}
cat <<EOF > "/data/data/com.termux/files/usr/var/lib/proot-distro/installed-rootfs/ubuntu/root/packinstall.sh"
packages=(${@:2})
EOF
cat <<'EOF' >> "/data/data/com.termux/files/usr/var/lib/proot-distro/installed-rootfs/ubuntu/root/packinstall.sh"
for package_name in "${packages[@]}"; do
desktop_files=$(dpkg-query -W -f='${binary:Package}\n' | grep "^$package_name\(-.*\)\?$" | xargs dpkg-query -L | grep "^/usr/share/applications/.*\.desktop$")
if [ -z "$desktop_files" ]; then
    echo "[1;31m[[0m-[1;31m][1;32m No .desktop files found for package [1;36m'$package_name' [1;32min/usr/share/applications."
else
    for desktop_files_name in $desktop_files; do
        desktop_files_with_ext=$(basename "$desktop_files_name")
        desktop_files_without_ext="${desktop_files_with_ext%.desktop}"
        echo "[1;31m[[0m-[1;31m][1;32m Adding [1;36m${desktop_files_without_ext} [1;32mTo Termux Menu"
        cp "/usr/share/applications/${desktop_files_with_ext}" "/data/data/com.termux/files/usr/share/applications/pd_added/"
        sed -i 's/Exec=/Exec=pdrun /g' "/data/data/com.termux/files/usr/share/applications/pd_added/${desktop_files_with_ext}"
    done
fi
done
rm packinstall.sh
EOF
proot-distro login ubuntu -- /bin/bash -c 'bash packinstall.sh'

elif [[ "$1" = "remove" ]] || [[ "$1" = "autoremove" ]]; then
proot-distro login --user  ubuntu --shared-tmp -- env DISPLAY=:1.0 sudo nala $1 ${@:2}
cat <<EOF > "/data/data/com.termux/files/usr/var/lib/proot-distro/installed-rootfs/ubuntu/root/packremove.sh"
packages=(${@:2})
EOF
cat <<'EOF' >> "/data/data/com.termux/files/usr/var/lib/proot-distro/installed-rootfs/ubuntu/root/packremove.sh"
for package_name in "${packages[@]}"; do
desktop_files=$(dpkg-query -W -f='${binary:Package}\n' | grep "^$package_name\(-.*\)\?$" | xargs dpkg-query -L | grep "^/usr/share/applications/.*\.desktop$")
if [ -z "$desktop_files" ]; then
    echo "[1;31m[[0m-[1;31m][1;32m No .desktop files found for package [1;36m'$package_name' [1;32min/usr/share/applications."
else
    for desktop_files_name in $desktop_files; do
        desktop_files_with_ext=$(basename "$desktop_files_name")
        desktop_files_without_ext="${desktop_files_with_ext%.desktop}"
        echo "[1;31m[[0m-[1;31m][1;32m Removing [1;36m${desktop_files_without_ext} [1;32mFrom Termux Menu" 
        rm /data/data/com.termux/files/usr/share/applications/pd_added/${desktop_files_with_ext}
    done
fi
done
rm packremove.sh
EOF
proot-distro login ubuntu -- /bin/bash -c 'bash packremove.sh'
elif [[ "$1" = "update" ]]; then
proot-distro login --user  ubuntu --shared-tmp -- env DISPLAY=:1.0 sudo nala update
fi

if [[ $1 = "install" ]] || [[ $1 = "remove" ]] || [[ $1 = "autoremove" ]]; then
gtk-update-icon-cache
update-desktop-database -q "/data/data/com.termux/files/usr/share/applications/pd_added"
fi

function remove_distro_container() {
     while true; do
        read -p "[1;31m[[0m-[1;31m][1;33m[1m Do you want to remove the distro[1;31m(it can't be undone) [1;33m(y/n) [0m" ans_remove_distro
        case $ans_remove_distro in
            [yY]* )
                echo "[1;31m[[0m-[1;31m][1;32mContinuing with answer: $ans_remove_distro[0m"
                sleep 0.2
                if [[ -d /data/data/com.termux/files/usr/var/lib/proot-distro/installed-rootfs/ubuntu ]]; then
                proot-distro remove ubuntu
                distro_file_list=(
                    "/data/data/com.termux/files/usr/bin/pdrun"
                    "/data/data/com.termux/files/usr/share/applications/pd_added"
                    "/data/data/com.termux/files/usr/bin/add2menu"
                    "/data/data/com.termux/files/usr/share/applications/add2menu.desktop"
                    "/data/data/com.termux/files/usr/bin/ubuntu"
                )
                    for distro_file in ""; do
                    if [[ -e "" ]]; then
                        if [[ -d "" ]]; then
                        rm -rf 
                        else
                        rm 
                        fi
                    else
                    echo "[1;31mFile does not exist: "[0m
                    fi
                    done
                echo "[1;32m6473selected_distro removed successfully"[0m
                echo "distro_add_answer=\"n\"" >> /data/data/com.termux/files/usr/etc/termux-desktop/configuration
                else
                echo "[1;31mubuntu does not exist"
                fi
                break;;
            [nN]* )
                echo "[1;31m[[0m-[1;31m][1;36mSkipping distro remove"[0m
                sleep 0.2
                break;;
            * )
                echo "[1;31m[[0m-[1;31m][1;31mInvalid input. Please enter 'y' or 'n'."[0m
                ;;
        esac
    done
}

function help_distro_container() {
echo -e "
ubuntu install packagename     to install packages and add them to termux desktop\n
ubuntu remove packagename      to remove packages and also remove them to termux desktop\n
ubuntu update                  to update the distro\n
ubuntu -r                      to login as root\n
ubuntu --remove                to complitely remove the distro and delete all files and data related to it\n
"
}
case $1 in
--root|-r)
proot-distro sh ubuntu
;;
--remove)
remove_distro_container
;;
--help|-h)
help_distro_container
;;
esac

RimaStudioo, 2024-11-13:
Nice, thanks very much

closed 2024-11-13 by RimaStudioo