sabamdarif/termux-desktop #43: Audio problem

RimaStudioo, 2024-11-13:
I've seen some comments on your YouTube, but I have problems with the audio (the audio doesn't work) even though I'm already using Firefox and trying to run Firefox using pdrun, and I'm using TX11. graphics Freedreno and Turnip

sabamdarif, 2024-11-13:
through out the setup you get question like this
_Do you want to configure audio support for distro container_ did you type `y` or not ?

sabamdarif, 2024-11-13:
and also is audio working in native termux ?

RimaStudioo, 2024-11-13:
> through out the setup you get question like this _Do you want to configure audio support for distro container_ did you type `y` or not ?

i choose yes, i've check on terminal to run pulse audiio, the result like this

pulseaudio                                                              1 ↵
W: [pulseaudio] pid.c: Stale PID file, overwriting.
E: [null-sink] util.c: Failed to connect to system bus: Failed to connect to socket /data/data/com.termux/files/usr/var/run/dbus/system_bus_socket: No such file or directory
E: [null-sink] util.c: Failed to connect to system bus: Failed to connect to socket /data/data/com.termux/files/usr/var/run/dbus/system_bus_socket: No such file or directory
E: [null-sink] util.c: Failed to connect to system bus: Failed to connect to socket /data/data/com.termux/files/usr/var/run/dbus/system_bus_socket: No such file or directory
E: [null-sink] util.c: Failed to connect to system bus: Failed to connect to socket /data/data/com.termux/files/usr/var/run/dbus/system_bus_socket: No such file or directory
E: [null-sink] util.c: Failed to connect to system bus: Failed to connect to socket /data/data/com.termux/files/usr/var/run/dbus/system_bus_socket: No such file or directory
E: [pulseaudio] module-console-kit.c: Unable to contact D-Bus system bus: org.freedesktop.DBus.Error.FileNotFound: Failed to connect to socket /data/data/com.termux/files/usr/var/run/dbus/system_bus_socket: No such file or directory
E: [pulseaudio] module.c: Failed to load module "module-console-kit" (argument: ""): initialization failed.
W: [pulseaudio] cli-command.c: stat('/data/data/com.termux/files/usr/etc/pulse/default.pa.d'): Unknown error 2
E: [pulseaudio] ltdl-bind-now.c: Failed to open module /data/data/com.termux/files/usr/lib/pulseaudio/modules/module-sles-sink.so: dlopen failed: library "libmediadrmmetrics_lite.so" not found: needed by /system/lib64/libmediadrm.so in namespace (default)
E: [pulseaudio] module.c: Failed to open module "module-sles-sink".

RimaStudioo, 2024-11-13:
> and also is audio working in native termux ?

wait i'll try it

RimaStudioo, 2024-11-13:
> and also is audio working in native termux ?

sorry ! how to test audio on native termux ?

RimaStudioo, 2024-11-13:
when i run pdrun firexfox from terminal, result like this

GetShaderSource() ->

#extension GL_EXT_shader_texture_lod: require
void main() {}

JavaScript warning: https://www.google.com/js/th/62mh7a0fQTK9Uep7g0y3snI_COPB4Zut5ZKHWl7ffDc.js line 2 > eval line 1483 > eval line 1 > eval line 1 > eval, line 1: WEBGL_debug_renderer_info is deprecated in Firefox and will be removed. Please use RENDERER.
[Child 23999, MediaDecoderStateMachine #1] WARNING: b400006faef0c080 Can't get cubeb context!: file /home/runner/.termux-build/firefox/src/dom/media/AudioStream.cpp:255
[Child 23999, MediaDecoderStateMachine #1] WARNING: Decoder=b400006fae320500 [OnMediaSinkAudioError]: file /home/runner/.termux-build/firefox/src/dom/media/MediaDecoderStateMachine.cpp:4626

sabamdarif, 2024-11-13:
> > and also is audio working in native termux ?
> 
> sorry ! how to test audio on native termux ?

install firefox in termux `pkg install firefox -y` then play a youtube video

RimaStudioo, 2024-11-13:
> > > and also is audio working in native termux ?
> > 
> > 
> > sorry ! how to test audio on native termux ?
> 
> install firefox in termux `pkg install firefox -y` then play a youtube video

no, not working, i've try it

sabamdarif, 2024-11-13:
which device samsung ?

RimaStudioo, 2024-11-13:
> which device samsung ?

my device is huawei matepad 11 2021, snapdragon 865, andreno 650

sabamdarif, 2024-11-13:
apt reinstall pulseaudio -y

RimaStudioo, 2024-11-13:
> apt reinstall pulseaudio -y

the result is :
┌──(root@localhost)-[/home/Entity01]
# sudo apt reinstall pulseaudio -y
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 1,045 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://deb.debian.org/debian bookworm/main arm64 pulseaudio arm64 16.1+dfsg1-2+b1 [1,045 kB]
Fetched 1,045 kB in 1s (1,001 kB/s)        
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 55676 files and directories currently installed.)
Preparing to unpack .../pulseaudio_16.1+dfsg1-2+b1_arm64.deb ...
Unpacking pulseaudio (16.1+dfsg1-2+b1) over (16.1+dfsg1-2+b1) ...
dpkg: warning: symbolic link '/etc/alsa/conf.d/99-pulse.conf' size has changed f
rom 42 to 38
dpkg: warning: symbolic link '/etc/pulse/client.conf.d/01-enable-autospawn.conf'
 size has changed from 34 to 32
dpkg: warning: symbolic link '/lib/systemd/system/pulseaudio-enable-autospawn.se
rvice' size has changed from 18 to 9
Setting up pulseaudio (16.1+dfsg1-2+b1) ...
Failed to send reload request: No such file or directory
Failed to scan devices: Permission denied
invoke-rc.d: could not determine current runlevel
Processing triggers for man-db (2.11.2-2) ...
setpriv: setresgid failed: Operation not permitted
Processing triggers for dbus (1.14.10-1~deb12u1) ...

sabamdarif, 2024-11-13:
i mean in termux

RimaStudioo, 2024-11-14:
> i mean in termux

┌──(Entity01@localhost)-[~]
└─≽ apt reinstall pulseaudio -y
Usage: nala [OPTIONS] COMMAND [ARGS]...
Try 'nala -h' for help.
╭─ Error ────────────────────────────────────────────────────────────────────────────────╮
│ No such command 'reinstall'.                                                           │
╰────────────────────────────────────────────────────────────────────────────────────────╯

sabamdarif, 2024-11-14:
> ┌──(Entity01@localhost)-[~] └─≽ apt reinstall pulseaudio -y Usage: nala [OPTIONS] COMMAND [ARGS]... Try 'nala -h' for help. ╭─ Error [...]

first type 
`bash` 
then `apt reinstall pulseaudio -y`

RimaStudioo, 2024-11-14:
> first type `bash` then `apt reinstall pulseaudio -y`

yes, i've try it, but It still doesn't work, maybe it doesn't work on Huawei

sabamdarif, 2024-11-14:
> yes, i've try it, but It still doesn't work, maybe it doesn't work on Huawei

May be

closed 2024-11-18 by sabamdarif