sabamdarif/termux-desktop #44: Make a telegram or discord server for support, github is inconsistent.

Awashit, 2024-11-14:
Provide their links on this repo.

sabamdarif, 2024-11-14:
Ok i will create a telegram group soon

Awashit, 2024-11-14:
Umm hardware acceleration doesn't seem to work other than virgl on my mali device, is there any way to hardware acceleration with virgl zink with firefox?

sabamdarif, 2024-11-14:
did you tried Experimental GPU Drivers ?

Awashit, 2024-11-14:
Yeah i tried those too is there some code i need to run before in order to enable hardware acceleration in firefox and chromium

Awashit, 2024-11-14:
My device is samsung tablet a9
Its pretty mid

sabamdarif, 2024-11-14:
> Yeah i tried those too is there some code i need to run before in order to enable hardware acceleration in firefox and chromium

enable webgl

Awashit, 2024-11-14:
Did it work for you on your mali device
Do i need to enable webgl.disabled to false and webgl force enabled to true in about config?
I will try today and inform you situation

Awashit, 2024-11-15:
[image]
[image]
[image]
[image]
[image]

Awashit, 2024-11-15:
Virgl-zink too didnt work

Awashit, 2024-11-15:
Only virgl seems to work

sabamdarif, 2024-11-15:
Experimental GPU Drivers have only support for vulkan thats why it didn't work

Awashit, 2024-11-17:
Currently no hardware acceleration works including viral
It seems they only work on lxqt

sabamdarif, 2024-11-17:
https://t.me/hello_android_0 join me here

closed 2024-11-18 by sabamdarif