sabamdarif/termux-desktop #47: Error pdrun libreoffice

KimetsuAndrea, 2024-11-19:
Error: got excessive positional argument 'env'. Note that distribution can be specified only once.

Usage: proot-distro login [OPTIONS] [DISTRO ALIAS] [-- [COMMAND]]

Command aliases: sh

Launch a login shell for the specified distribution if no
additional arguments were given. Otherwise execute the
given command and exit.

Options:

  --help               - Show this help information.

  --user [user]        - Login as specified user instead of 'root'.

  --fix-low-ports      - Modify bindings to protected ports to use
                         a higher port number.

  --isolated           - Run isolated environment without access
                         to host file system.

  --termux-home        - Mount Termux home directory to /root.
                         Takes priority over '--isolated' option.

  --shared-tmp         - Mount Termux temp directory to /tmp.
                         Takes priority over '--isolated' option.

  --bind [path:path]   - Custom file system binding. Can be specified
                         multiple times.
                         Takes priority over '--isolated' option.

  --no-link2symlink    - Disable hardlink emulation by proot.
                         Adviseable only on devices with SELinux
                         in permissive mode.

  --no-sysvipc         - Disable System V IPC emulation by proot.

  --no-kill-on-exit    - Wait until all running processes will finish
                         before exiting. This will cause proot to
                         freeze if you are running daemons.

  --kernel [string]    - Set the kernel release and compatibility
                         level to string.

  --work-dir [path]    - Set the working directory.

  --env ENV=val        - Set environment variable. Can be specified
                         multiple times.

Put '--' if you wish to stop command line processing and pass
options as shell arguments.

If no '--isolated' option given, the following host directories
will be available:

  * /apex (only Android 10+)
  * /data/dalvik-cache
  * /data/data/com.termux
  * /sdcard
  * /storage
  * /system
  * /vendor

This should be enough to get Termux utilities like termux-api or
termux-open get working. If they do not work for some reason,
make sure they are properly set in /etc/environment.

Also check whether they define variables like ANDROID_DATA,
ANDROID_ROOT, BOOTCLASSPATH and others which are usually set
in Termux sessions.

If issue occurs only after su/sudo use, then likely your PAM
configuration doesn't load /etc/environment and you need to fix
it by enabling pam_env.so in /etc/pam.d configuration.

Example PAM configuration line:

  session  required  pam_env.so readenv=1

You need to append it to /etc/pam.d/su, /etc/pam.d/sudo or other
file depending on distribution.

Selected distribution should be referenced by alias which can be
obtained by this command: proot-distro list

Proot-Distro v4.17.2 by Termux (@sylirre).

KimetsuAndrea, 2024-11-19:
Hi, can you Add a retrying System, my connection Aborted, while installing Arch Destro.

KimetsuAndrea, 2024-11-19:
▀█▀ █▀▀ █▀█ █▀▄▀█ █░█ ▀▄▀   █▀▄ █▀▀ █▀ █▄▀ ▀█▀ █▀█ █▀█
 ░█░ ██▄ █▀▄ █░▀░█ █▄█ █░█   █▄▀ ██▄ ▄█ █░█ ░█░ █▄█ █▀▀

 Install Gui Desktop In Termux
 code by @sabamdrif

┌──(Delfin@localhost)-[~]
└─≽ setup-termux-desktop --reinstall config
[-] Checking for update...

[-] Good job, you are using the latest installer
empty (sub)expression/data/data/com.termux/files/usr/etc/termux-desktop/configuration: line 29: xz-utils=already_exist: command not found
/data/data/com.termux/files/usr/etc/termux-desktop/configuration: line 32: x11-repo=already_exist: command not found
/data/data/com.termux/files/usr/etc/termux-desktop/configuration: line 33: tur-repo=already_exist: command not found

sabamdarif, 2024-11-20:
> > > Hi, can you Add a retrying System, my connection Aborted, while installing Arch Destro.
> > 
> > 
> > is the connection problem in termux or in arch linux ( when it tried to update or install any package )
> 
> From my internet 😭

Use vpn

commit b350439d (2024-11-20): fix #47
closed 2024-11-20 by sabamdarif