sabamdarif/termux-desktop #50: Chinese and the input method for Chinese

2061360308, 2024-11-23:
How can I set the system language to Chinese and how can I add a Chinese input method? I want to use fcitx.

darkwebdisclaimer, 2024-12-03:
### `how to fix issues my firefox keep on closing by it self`

# my android detiels

* System Info:
* System          :   Android 12
* Host            :   Itel itel A662L
* Kernel          :   Linux 5.4.233+-ab701
* CPU             :   Spreadtrum SC9832E (4 vCPU)
* Termux Version  :   0.119.0-beta.1-
* Memory          :   909Mi used, 1.9Gi total
* Temperature     :   °c

sabamdarif, 2024-12-03:
> ### `how to fix issues my firefox keep on closing by it self`
> # my android detiels
>
>     * System Info:
>
>     * System          :   Android 12
>
>     * Host            :   Itel itel A662L
>
>     * [...]

might be gpu issue / ram issue

sabamdarif, 2024-12-20:
> How can I set the system language to Chinese and how can I add a Chinese input method? I want to use fcitx.

https://ivonblog.com/en-us/posts/termux-x11/#7-how-to-type-chinse-in-termux-x11

closed 2024-12-20 by sabamdarif