sabamdarif/termux-desktop #55: Error interpretation JPEG image file

clifordunique, 2024-12-15:
Trying to use ristretto and even thunar to open image I received this error. It says (wrong JPEG library version: library is 62, caller expects 80). 
This error causes a problem that I can't even see the download wallpapers

sabamdarif, 2024-12-15:
pkg reinstall libjpeg-turbo
pkg install libjpeg-turbo-progs libjpeg-turbo-static

clifordunique, 2024-12-15:
Already reinstall it but no change

clifordunique, 2024-12-15:
I have Uninstall termux and do a clean installation of termux but the same problem persist. With JPEG library version.

sabamdarif, 2024-12-15:
then you should ask it on termux's group

closed 2024-12-20 by sabamdarif

lenquan, 2025-02-22:
I wanted to comment I'm experiencing the same thing. I've done several clean installs testing different xfce options on the script. I encounter different issues with different paths.

When going custom path using minimal look and the mesa-vulkan-icd-wrapper, the extended right click options and pulse audio breaks, BUT the jpeg images is resolved as it looks like the symbolic link from jpeg62 to jpeg 8 works.

However, whenever I install the general path from your first option with HW acceleration. This breaks the JPEG option, but fixes the pulse audio issue. So I think there's a disconnect there when you go custom which breaks pulse audio and the general option which breaks jpeg images @sabamdarif which I don't think reaching out to termux would resolve if it works on some reinstall instances but not others.

sabamdarif, 2025-02-22:
@lenquan can you install all packages related to xfce and see if it's fixed not, run
```bash
apt install xfce4* -y
```

lenquan, 2025-02-22:
> [@lenquan](https://github.com/lenquan) can you install all packages related to xfce and see if it's fixed not, run
> 
> apt install xfce4* -y

@sabamdarif, ran this on a new install following the generic option 1 with HW acceleration. The JPEG still is having error. I'm going to attempt the custom installation with minimal but follow majority of the setting of the generic option to see if this resolves the libjpeg error, but maintains pulse audio functionality.