sabamdarif/termux-desktop #60: starting the VNC server and X11

GoudarziMehdi, 2025-01-05:
[image]
[image]
Problem starting the VNC server and X11:
The related Perl scripts have syntax errors.

sabamdarif, 2025-01-05:
it just fixed few hours ago try again

GoudarziMehdi, 2025-01-05:
The first screenshot is from an hour ago

sabamdarif, 2025-01-05:
cd $PREFIX/bin
cat vncstart

And send the text output here
 also type tdconfig and send that too

NotAl2, 2025-01-05:
tried the script yesterday was working fine reinstalled today and got the same error. manually editing the tx11start file snd removing all such '&' did seem to work but i want to know if those '&' were crucial for smt and so will it break anything if i remove them?

NotAl2, 2025-01-05:
tried the script yesterday was working fine reinstalled today and got the same error. manually editing the tx11start file snd removing all such '&' did seem to work but i want to know if those '&' were crucial for smt and so will it break anything if i remove them?

**Here is the cat result:**

`.../usr/bin $ cat tx11start
#!/data/data/com.termux/files/usr/bin/bash

termux-wake-lock

pulseaudio --start --exit-idle-time=-1

if [[ "$1" == "--debug" ]]; then
case $2 in
--nogpu)
    # Start Termux X11 without GPU acceleration
    XDG_RUNTIME_DIR=${TMPDIR} termux-x11 :0 &
    sleep 1
    am start --user 0 -n com.termux.x11/com.termux.x11.MainActivity > /dev/null 2>&1
    sleep 1
    env DISPLAY=:0 XDG_CONFIG_DIRS=/data/data/com.termux/files/usr/etc/xdg  LIBGL_ALWAYS_SOFTWARE=1 MESA_LOADER_DRIVER_OVERRIDE=llvmpipe GALLIUM_DRIVER=llvmpipe dbus-launch --exit-with-session xfce4-session

    # Check if the second argument is --legacy
    if [[ "$3" == "--legacy" ]]; then
        XDG_RUNTIME_DIR=${TMPDIR} termux-x11 :0 -legacy-drawing &
        sleep 1
        am start --user 0 -n com.termux.x11/com.termux.x11.MainActivity > /dev/null 2>&1
        sleep 1
        env DISPLAY=:0 XDG_CONFIG_DIRS=/data/data/com.termux/files/usr/etc/xdg  dbus-launch --exit-with-session xfce4-session
    fi
    ;;

--nodbus)
    # Start Termux X11 without dbus-launch
        export
         &
    env XDG_RUNTIME_DIR=${TMPDIR} XDG_CONFIG_DIRS=/data/data/com.termux/files/usr/etc/xdg   termux-x11 :0 -xstartup xfce4-session
    exit 0

    # Nested case to check for additional options
    case $3 in
    --nogpu)
        env LIBGL_ALWAYS_SOFTWARE=1 MESA_LOADER_DRIVER_OVERRIDE=llvmpipe GALLIUM_DRIVER=llvmpipe XDG_RUNTIME_DIR=${TMPDIR} XDG_CONFIG_DIRS=/data/data/com.termux/files/usr/etc/xdg  termux-x11 :0 -xstartup xfce4-session

        if [[ "$4" == "--legacy" ]]; then
            env XDG_RUNTIME_DIR=${TMPDIR} XDG_CONFIG_DIRS=/data/data/com.termux/files/usr/etc/xdg  termux-x11 :0 -legacy-drawing -xstartup xfce4-session
        fi
        ;;
    --legacy)
        export
         &
        env XDG_RUNTIME_DIR=${TMPDIR} XDG_CONFIG_DIRS=/data/data/com.termux/files/usr/etc/xdg   termux-x11 :0 -legacy-drawing -xstartup xfce4-session
        ;;
    *)
        echo -e "--legacy to start termux:x11 with -legacy-drawing"
        echo -e "--nogpu to start termux:x11 without GPU acceleration"
        exit 0
        ;;
    esac
    ;;

--legacy)
    # Start Termux X11 with legacy drawing mode
        export
         &
    sleep 1
    XDG_RUNTIME_DIR=${TMPDIR} termux-x11 :0 -legacy-drawing &
    sleep 1
    am start --user 0 -n com.termux.x11/com.termux.x11.MainActivity > /dev/null 2>&1 &
    sleep 1
    env DISPLAY=:0 XDG_CONFIG_DIRS=/data/data/com.termux/files/usr/etc/xdg   dbus-launch --exit-with-session xfce4-session
[... 79 lines cut ...]
    sleep 1
    XDG_RUNTIME_DIR=${TMPDIR} termux-x11 :0 -legacy-drawing &
    sleep 1
    am start --user 0 -n com.termux.x11/com.termux.x11.MainActivity > /dev/null 2>&1 &
    sleep 1
    env DISPLAY=:0 XDG_CONFIG_DIRS=/data/data/com.termux/files/usr/etc/xdg   dbus-launch --exit-with-session xfce4-session > /dev/null 2>&1 &
    ;;
*)
    # Default behavior: start Termux X11 with GPU acceleration and dbus
        export
         &
    sleep 1
    XDG_RUNTIME_DIR=${TMPDIR} termux-x11 :0 &
    sleep 1
    am start --user 0 -n com.termux.x11/com.termux.x11.MainActivity > /dev/null 2>&1 &
    sleep 1
    env DISPLAY=:0 XDG_CONFIG_DIRS=/data/data/com.termux/files/usr/etc/xdg   dbus-launch --exit-with-session xfce4-session > /dev/null 2>&1 &
    ;;
esac
fi
sleep 5
process_id=$(ps -aux | grep '[x]fce4-screensaver' | awk '{print $2}')
kill "$process_id" > /dev/null 2>&1 `

and tdconfig didn't work for me

NotAl2, 2025-01-05:
i just refined the tx11start script using (credits to chatgpt) and it seems to work fine. replacing the original start script present in /data/data/com.termux/files/usr/bin/tx11start with this one should work.

Refined Script:
#!/data/data/com.termux/files/usr/bin/bash

# Keep the device awake
termux-wake-lock

# Start Pulseaudio
pulseaudio --start --exit-idle-time=-1

# Function to display help
display_help() {
    echo -e "\e[1;36mUsage: tx11start [OPTION]\e[0m"
    echo -e "\e[1;32mOptions:\e[0m"
    echo -e "  --nogpu       Start Termux X11 without GPU acceleration"
    echo -e "  --nodbus      Start Termux X11 without dbus-launch"
    echo -e "  --legacy      Start Termux X11 with legacy drawing mode"
    echo -e "  --debug       Enable debug logs"
    echo -e "  --help        Display this help message"
    exit 0
}

# Main execution
case $1 in
--nogpu)
    echo "Starting Termux X11 without GPU acceleration..."
    export LIBGL_ALWAYS_SOFTWARE=1
    export MESA_LOADER_DRIVER_OVERRIDE=llvmpipe
    export GALLIUM_DRIVER=llvmpipe
    XDG_RUNTIME_DIR=${TMPDIR} termux-x11 :0 &
    sleep 1
    am start --user 0 -n com.termux.x11/com.termux.x11.MainActivity > /dev/null 2>&1
    env DISPLAY=:0 XDG_CONFIG_DIRS=/data/data/com.termux/files/usr/etc/xdg dbus-launch --exit-with-session xfce4-session
    ;;
--nodbus)
    echo "Starting Termux X11 without dbus-launch..."
    export XDG_RUNTIME_DIR=${TMPDIR}
    export XDG_CONFIG_DIRS=/data/data/com.termux/files/usr/etc/xdg
    termux-x11 :0 -xstartup xfce4-session &
    ;;
--legacy)
    echo "Starting Termux X11 with legacy drawing mode..."
    export XDG_RUNTIME_DIR=${TMPDIR}
    export XDG_CONFIG_DIRS=/data/data/com.termux/files/usr/etc/xdg
    termux-x11 :0 -legacy-drawing -xstartup xfce4-session &
    ;;
--debug)
    echo "Debug mode enabled. Logs will be displayed."
    set -x
    ;;
--help)
    display_help
    ;;
*)
    echo "Starting Termux X11 with default settings..."
    XDG_RUNTIME_DIR=${TMPDIR} termux-x11 :0 &
    sleep 1
    am start --user 0 -n com.termux.x11/com.termux.x11.MainActivity > /dev/null 2>&1
    env DISPLAY=:0 XDG_CONFIG_DIRS=/data/data/com.termux/files/usr/etc/xdg dbus-launch --exit-with-session xfce4-session
    ;;
esac

# Clean up unnecessary processes
sleep 5
process_id=$(pgrep -f 'xfce4-screensaver')
kill "$process_id" > /dev/null 2>&1

NotAl2, 2025-01-05:
i just refined the tx11start script using (credits to chatgpt) and it seems to work fine. replacing the contents of the original start script present in /data/data/com.termux/files/usr/bin/tx11start with this one should work.
[tx11start refined script.txt](https://github.com/user-attachments/files/18309931/tx11start.refined.script.txt)

sabamdarif, 2025-01-05:
if tdconfig didn't work that mean the install failed

NotAl2, 2025-01-05:
have you fixed the issue yet?
also yesterday it was working just fine

commit 1b2e2fd2 (2025-01-05): fix #60
closed 2025-01-05 by sabamdarif