sabamdarif/termux-desktop #61: The problem persists, along with a couple of other problems

GoudarziMehdi, 2025-01-05:
[image]
[image]
[image]
[image]
[image]
I tested it on both Ubuntu and Debian

sabamdarif, 2025-01-06:
type tdconfig and send everything it print

GoudarziMehdi, 2025-01-06:
~ $ tdconfig
####################################
########## Termux Desktop ##########
####################################

########################
#  -:Device Details:-  #
########################
#
# Termux Version: googleplay.2024.10.30-
# Device Model: Xiaomi 21081111RG
# Android Version: 14
# Total Ram: 7.3Gi
# Architecture: aarch64
# System CPU Architecture: arm64-v8a,armeabi-v7a,armeabi
# Processor: Mediatek MT6893
# Network Condition: Unknown,Unknown (On first run)
# Country: ir
#
########################

##### Please don't modify this file otherwise some functions won't work #####

"wget"=already_exist
"pv"=already_exist
"jq"=already_exist
"curl"=already_exist
"pulseaudio"=already_exist
"tar"=already_exist
"gzip"=already_exist
de_startup=xfce4-session
de_name=xfce
themes_folder=/data/data/com.termux/files/home/.themes
icons_folder=/data/data/com.termux/files/home/.icons
style_answer=1
style_name=Basic Style
distro_add_answer=y
confirmation_mesa_vulkan_icd_wrapper=y
distro_answer=2
selected_distro=ubuntu
distro_zsh_answer=n
distro_terminal_utility_setup_answer=n
pd_useradd_answer=y
pd_pass_type=2
user_name=dev
pass=1
device_gpu_model=2
device_gpu_model_name=mali
exp_termux_gl_hw_answer_num=1
exp_termux_gl_hw_answer=zink
termux_hw_answer=zink
pd_hw_answer_num=1
pd_hw_answer=zink
zsh_answer=n
"xfce4"=already_exist
"kvantum"=already_exist
"xwayland"=already_exist
"pavucontrol"=already_exist
"atril"=already_exist
"galculator"=already_exist
installed_browser=skip
installed_ide=skip
installed_media_player=skip
installed_photo_editor=skip
install_synaptic=n
setup_wine=skip
pd_audio_config_answer=y
final_user_name=dev
"clvk"=already_exist
pd_shell_name=bash
"zenity"=already_exist
distro_add_answer=y
"tigervnc"=already_exist
gui_mode=both
terminal_utility_setup_answer=n
de_on_startup=y

sabamdarif, 2025-01-06:
it look like you have a internet connection problem, use a stable connection and if possible use reliable vpn

closed 2025-01-18 by sabamdarif