sabamdarif/termux-desktop #62: Ubuntu container no longer works?

InfamousKnight, 2025-01-07:
I looked under container code, and it appears ubuntu doesn't have any safe guards to make sure it installs..

I need ubuntu for fex-emu. Lately it hasn't been installing correctly.

I have nordvpn for my VPN.

sabamdarif, 2025-01-08:
can you not login into ubuntu ?

InfamousKnight, 2025-01-08:
Well, I could with proot-distro login.

But for some reason fex doesn't install anywhere. I even tried doing it the
old fashioned way

sabamdarif, 2025-01-08:
> Well, I could with proot-distro login.
try login with ubuntu command and tell me what it shows
and for fex if it don't install then can't anything about it

InfamousKnight, 2025-01-08:
It might actually be installed correctly and I just didn't realize it.

Does fex run automatically? I thought it had to be used with fex [insert
application] like box64.

This is what happens whenever I install a library:
Setting up fex-emu-binfmt32 (2501~n) ...
systemd-binfmt: unrecognized service
dpkg: error processing package fex-emu-binfmt32 (--configure):
installed fex-emu-binfmt32 package post-installation script subprocess
returned
error exit status 1
Setting up fex-emu-binfmt64 (2501~n) ...
systemd-binfmt: unrecognized service
dpkg: error processing package fex-emu-binfmt64 (--configure):
installed fex-emu-binfmt64 package post-installation script subprocess
returned
error exit status 1
Errors were encountered while processing:
fex-emu-binfmt32
fex-emu-binfmt64
E: Sub-process /usr/bin/dpkg returned an error code (1)

So i guess it's there?

And when I try to run bizhawk, this shows up:

MESA: error: ZINK: vkCreateInstance failed (VK_ERROR_INCOMPATIBLE_DRIVER)
glx: failed to create drisw screen

Looked that up online and it says it's a ubuntu 24.10 problem and to
downgrade to ubuntu 22

sabamdarif, 2025-01-08:
i never use fex it look like some kind of systmd issue

and the second one is that is have issue zink change it to something else

closed 2025-01-10 by sabamdarif

InfamousKnight, 2025-01-12:
I just read online that fex support has been dropped thanks to overprotective android garbage.

Cant wait for android 16 to come out with full linux emulator!