sabamdarif/termux-desktop #70: Hello! Help solve the problem.

Foum02, 2025-02-02:
My problem is that when I enter the terminal in a launched Gui, I have Termux:x11 automatically start again and the Gui is disabled after startup. 
Please help!

NickoDPu, 2025-02-03:
Idk about termux x11 but are you have vncserver? Because I knew how to solve vncserver problems.

sabamdarif, 2025-02-03:
> My problem is that when I enter the terminal in a launched Gui, I have Termux:x11 automatically start again and the Gui is disabled after startup. Please help!

it's because you enable auto start, i see this problem before in some phones where for some reason it can't detect the existing x11 process.
i can fix that if you do the thing i will ask, will you ?

Foum02, 2025-02-03:
> > Моя проблема в том, что когда я ввожу команду в терминале запущенного графического интерфейса, Termux:x11 автоматически запускается снова, а графический интерфейс отключается после запуска. Пожалуйста, помогите!
> 
> это из-за того, что вы включили автозапуск. Я уже сталкивался с этой проблемой на некоторых телефонах, где по какой-то причине не удаётся обнаружить существующий процесс x11. Я могу это исправить, если вы сделаете то, о чём я вас попрошу. Вы согласны?

Yes I agree

sabamdarif, 2025-02-03:
```bash
setup-termux-desktop --change autostart 
```
run this removed the autostart

sabamdarif, 2025-02-03:
for those who have this issue open termux then it will start tye de, go back to termux type top then zoom out , see there will be a process which will have the termux x11 word at the end, send me what it shows i that line

queso42, 2025-02-03:
Well, it detect the termux-x11 process port wrong, it should be :0 but it expects :1

Foum02, 2025-02-04:
> для тех, у кого возникла эта проблема, откройте Termux, затем введите «top», затем уменьшите масштаб, увидите процесс, в конце которого будет слово «Termux x11», отправьте мне то, что отображается в этой строке

Thank you very much

sabamdarif, 2025-02-04:
> Well, it detect the termux-x11 process port wrong, it should be :0 but it expects :1

yes you are right i miss that, i will fix it soon

sabamdarif, 2025-02-04:
> > для тех, у кого возникла эта проблема, откройте Termux, затем введите «top», затем уменьшите масштаб, увидите процесс, в конце которого будет слово «Termux x11», отправьте мне то, что отображается в этой строке
> 
> 17110 u0_a400 10 -10 4.6G 71M 70M S 2.3 1.9 0:05.01 app_process -Xnoimage-dex2oat / com.termux.x11.Loader+
> 
> You are talking about this line

ok it will be fixed

Foum02, 2025-02-04:
> настройка-termux-рабочий стол --изменить автозапуск 
> запустите это, удалив автозапуск

Почему-то на в MacOS стиль это не работает

sabamdarif, 2025-02-04:
type nano .zshrc (for zsh ) or /data/data/com.termux/files/usr/etc/bash.bashrc (bash) 
remove 
remove the lines from 
# Start Termux:X11
to 
fi

then save and exit

Foum02, 2025-02-04:
> введите nano .zshrc (для zsh) или /data/data/com.termux/files/usr/etc/bash.bashrc (bash) удалите удалите строки из "# Запустите Termux:X11 до fi"
> 
> затем сохраните и выйдите

Understood, thank you

closed 2025-02-09 by Foum02