sabamdarif/termux-desktop #75: Tunar file manager problem

mdshovon97, 2025-02-09:
Tunar file manager > File system> folder, file like etc, lib all not found.

sabamdarif, 2025-02-09:
"File system" it's only access able if you have root

mdshovon97, 2025-02-10:
**Bug fix and improve please**

# System sound pulseaudio setting not   working but sound working (system try    option not showing)

# Please update Graphic driver Qualcomm 6xx and others driver

# Do not show file system root folder        and any files.
- Others distro show thunar file system root folder without root but here is not showing.

# pkg command not working
for example pkg install.. pkg update.

I hope you know about this. Please update and fix thank you. This termux distro is best because device gpu driver properly work without root, You just try to improve this some features. Thank you.

sabamdarif, 2025-02-10:
> System sound pulseaudio setting not working but sound working (system try option not showing)

device name ?
also in termux type `tdconfig` and send me all the text it shows, and is all apps have no sound or just the browsers ?

> Please update Graphic driver Qualcomm 6xx and others driver

exactly what issue are facing right now ?

> Do not show file system root folder and any files.
> Others distro show thunar file system root folder without root but here is not showing.

that because the other distro you use they are run using proot with their own filesystem, and in the case of termux native there root folder mean your phone's root filesystem which you can't access without actual root

>pkg command not working
>for example pkg install.. pkg update.

run pkg install.. / pkg update and send me the error it's shows / screenshot of what happening

mdshovon97, 2025-02-10:
Bro others xfce atlist showing all file and folder (root folder) but here is empty invisable. Plz try to more improve tnx. My secoundry device realme 7 pro Qc 720G adreno 618 > "File system" it's only access able if you have root.

mdshovon97, 2025-02-11:
Maybe edit GNU nano file
How to edit gnu nano file user All All All like this.
> Bro others xfce atlist showing all file and folder (root folder) but here is empty invisable. Plz try to more improve tnx. My secoundry device realme 7 pro Qc 720G adreno 618 > "File system" it's only access able if you have root.
> 
>

mdshovon97, 2025-02-11:
It will be used to enable opengl support

1. Vulkan (ZINK)

2. OpenGL (VIRGL ANGLE)
                                                     3. Vulkan (VIRGL ANGLE)

4. OpenGL ES (ZINK VIRGL)

5. The Vulkan-Icd-Wrapper Driver With Mesa

6. The Vulkan-Icd-Wrapper Driver With Mesa-Zink

7. Freedreno KGSL (Unstable | Adreno GPU Only)
What is best for my device SD720G Adreno 618
> "File system" it's only access able if you have root

sabamdarif, 2025-02-16:
> What is best for my device SD720G Adreno 618

try freedreno if it didn't word then then 5 
also for freedreno type `n` to icd-wrapper question then select freedreno

closed 2025-03-10 by sabamdarif