sabamdarif/termux-desktop #76: VNC error

NammIsADev, 2025-02-10:
Logs: (Vulkan)
Could not start Xvnc.

Xvnc TigerVNC 1.14.1 - built Oct 29 2024 03:31:59
Copyright (C) 1999-2024 TigerVNC Team and many others (see README.rst)
See https://www.tigervnc.org for information on TigerVNC.
Underlying X server release 12101008

Built without GBM support
Failed to initialize DRI3 extension
Sync Extension 3.1
(EE) Segmentation fault at address 0x0
(EE)
Fatal server error:
(EE) Caught signal 11 (Segmentation fault). Server aborting
(EE)

GPU I'm using: Mali T830 MP2
Render method I tried: Vulkan (Zink), OpenGL (VIRGL ANGLE), OpenGL ES (Zink VIRGL)
Reinstall times: 3 times

Another logs but on OpenGL/OpenGL ES:
client: VTEST_CLIENT_DISCONNECTED
SIGSEGV! 
Could not start Xvnc. 
Xvnc TigerVNC 1.14.1 - built Oct 29 2024 03:31:59 
Copyright (C) 1999-2024 TigerVNC Team and many others (see README.rst) 
See https://www.tigervnc.org for information on TigerVNC. 
Underlying X server release 12101008 
Built without GBM support 
Failed to initialize DRI3 extension 
Sync Extension 3.1 
lost connection to rendering server on 8 read -1 104 
(EE) Fatal server error: 
(EE) Caught signal 6 (Aborted). Server aborting 
(EE)

System Compatibilty Check said:

[image]

Device using: Huawei Nova 2i
Termux version: Latest

sabamdarif, 2025-02-10:
where the other parts of `tdconfig` also the GPU section is blank which gpu you have ?

sabamdarif, 2025-02-11:
> > other parts of `tdconfig`
> 
> doesn't have, tdconfig only show that

then the install get  some issue during installation

NammIsADev, 2025-02-12:
> > > other parts of `tdconfig`
> > 
> > 
> > doesn't have, tdconfig only show that
> 
> then the install get some issue during installation

how to solve it

sabamdarif, 2025-02-12:
reinstall will be best, the tdconfig isn't a big thing to consider, but because it didn't right out that mean the install failed, at some point

NammIsADev, 2025-02-13:
> reinstall will be best, the tdconfig isn't a big thing to consider, but because it didn't right out that mean the install failed, at some point

i reinstalled 3 times

NammIsADev, 2025-02-13:
> did you tried termux x11

also failed

sabamdarif, 2025-02-14:
which install option you used, in the home folder there should a termux-desktop.log can you send that here

NammIsADev, 2025-02-14:
i deleted all installer files, will reinstall soon

sabamdarif, 2025-02-14:
> i deleted all installer files, will reinstall soon

do a clean install

NammIsADev, 2025-03-28:
closed because now i use chroot (my phone already rooted)

closed 2025-03-28 by NammIsADev