sabamdarif/termux-desktop #77: Hangover doesnt have desktop integration like mobox does? The wine file thingy?

gurachan, 2025-02-10:
How do i open hangover file manager easy as mobox does.

gurachan, 2025-02-10:
> it should have that if you can't see them, then run this
> 
> wget https://raw.githubusercontent.com/sabamdarif/termux-desktop/main/other/add-wine-shortcut
> chmod +x add-wine-shortcut
> ./add-wine-shortcut
> it will add those

./add-wine-shortcut: line 11: check_and_backup: command not found

gurachan, 2025-02-23:
```curl -Lf https://raw.githubusercontent.com/sabamdarif/termux-desktop/main/setup-termux-desktop -o setup-termux-desktop && chmod +x setup-termux-desktop && ./setup-termux-desktop```

Tried to reinstall it now it wont finish if i select hang over..

```shell

Unpacking hangover-wine (10.0-2) over (10.0-2) ...
Setting up hangover-wine (10.0-2) ...
[-] Processing package: winetricks
Checking availability of current mirror:
[*] https://packages-cf.termux.dev/apt/termux-main/: ok
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package winetricks
[✓] Adding wine desktop shortcuts...
https://raw.githubusercontent.com/sabamdarif/termux-desktop/main/other/add-wine-shortcut: No such file or directory
[☓] Failed to download the file, retrying...
https://raw.githubusercontent.com/sabamdarif/termux-desktop/main/other/add-wine-shortcut: No such file or directory
[☓] Failed to download the file after retry
~ $
```

sabamdarif, 2025-02-24:
you can add those manually 

```bash
#!/data/data/com.termux/files/usr/bin/bash

source /data/data/com.termux/files/usr/etc/termux-desktop/common_functions

function check_termux() {
	if [[ $HOME != *termux* ]]; then
	echo "${R}[${R}☓${R}]${R}${BOLD}Please run it inside termux${W}"
	exit 0
	fi
}

if [[ "$setup_wine" == "hangover-wine" ]]; then
wine_cmd="hangover-wine"
else
wine_cmd="wine"
fi

function wine_shortcut() {
rm -rf "$PREFIX/share/applications/wine-explorer.desktop $PREFIX/share/applications/wine-winecfg.desktop $PREFIX/share/applications/wine-uninstaller.desktop $PREFIX/share/applications/wine-regedit.desktop $PREFIX/share/applications/wine-wineboot.desktop" >/dev/null 2>&1

cat << EOF > "$PREFIX/share/applications/wine-explorer.desktop"
[Desktop Entry]
Name=Wine File
Comment=Wine File Browser
Name[ar]=مدير ملفات واين
Comment[ar]=إدارة الملفات بطريقة واين
Exec=${wine_cmd} explorer
Icon=winefile
Terminal=false
Type=Application
Categories=X-Wine;
X-Desktop-File-Install-Version=0.26
EOF
chmod +x "$PREFIX/share/applications/wine-explorer.desktop"
cat << EOF > "$PREFIX/share/applications/wine-winecfg.desktop"
[Desktop Entry]
Name=Wine Configuration
Comment=Interface to set wine parameters
Name[ar]=إعدادات واين
Comment[ar]=لوحة تحكم بمنظومة واين
Exec=${wine_cmd} winecfg
Icon=winecfg
Terminal=false
Type=Application
StartupNotify=true
StartupWMClass=winecfg.exe
Categories=X-Wine;
X-Desktop-File-Install-Version=0.26
EOF
chmod +x "$PREFIX/share/applications/wine-winecfg.desktop"
cat << EOF > "$PREFIX/share/applications/wine-uninstaller.desktop"
[Desktop Entry]
Name=Wine Software Uninstaller
Comment=Interface to uninstall software
Name[ar]=مزيل التطبيقات من واين
Comment[ar]=واجهة إزالة التّطبيقات من واين
Exec=${wine_cmd} uninstaller
Icon=msiexec
Terminal=false
Type=Application
StartupNotify=true
StartupWMClass=uninstaller.exe
Categories=X-Wine;
X-Desktop-File-Install-Version=0.26
EOF
chmod +x "$PREFIX/share/applications/wine-uninstaller.desktop"
cat << EOF > "$PREFIX/share/applications/wine-regedit.desktop"
[Desktop Entry]
Name=Regedit
Comment=Wine registry editor
Name[ar]=محرر السّجل
Comment[ar]=محرر لسجلات واين
Exec=${wine_cmd} regedit
Icon=regedit
Terminal=false
Type=Application
StartupNotify=true
StartupWMClass=regedit.exe
Categories=X-Wine;
X-Desktop-File-Install-Version=0.26
EOF
chmod +x "$PREFIX/share/applications/wine-regedit.desktop"
cat << EOF > "$PREFIX/share/applications/wine-wineboot.desktop"
[Desktop Entry]
Name=Wine Boot
Comment=Simulate restart
Name[ar]=إقلاع واين
Comment[ar]=محاكاة إعادة التشغيل لواين
Exec=${wine_cmd} wineboot
Icon=wine
Terminal=false
Type=Application
Categories=X-Wine;
X-Desktop-File-Install-Version=0.26
EOF
chmod +x "$PREFIX/share/applications/wine-wineboot.desktop"
}

check_termux
wine_shortcut
```

save it into a script and run

TomieAi, 2025-02-24:
> you can add those manually
> 
> #!/data/data/com.termux/files/usr/bin/bash
> 
> source /data/data/com.termux/files/usr/etc/termux-desktop/common_functions
> 
> function check_termux() {
> 	if [[ $HOME != *termux* ]]; then
> 	echo "${R}[${R}☓${R}]${R}${BOLD}Please run it inside termux${W}"
> 	exit 0
> 	fi
> }
> 
> if [[ "$setup_wine" == "hangover-wine" ]]; then
> wine_cmd="hangover-wine"
> else
> wine_cmd="wine"
> fi
> 
> function wine_shortcut() {
> rm -rf "$PREFIX/share/applications/wine-explorer.desktop $PREFIX/share/applications/wine-winecfg.desktop $PREFIX/share/applications/wine-uninstaller.desktop $PREFIX/share/applications/wine-regedit.desktop $PREFIX/share/applications/wine-wineboot.desktop" >/dev/null 2>&1
> 
> cat << EOF > "$PREFIX/share/applications/wine-explorer.desktop"
> [Desktop Entry]
> Name=Wine File
> Comment=Wine File Browser
> Name[ar]=مدير ملفات واين
> Comment[ar]=إدارة الملفات بطريقة واين
> Exec=${wine_cmd} explorer
> Icon=winefile
> Terminal=false
> Type=Application
> Categories=X-Wine;
> X-Desktop-File-Install-Version=0.26
> EOF
> chmod +x "$PREFIX/share/applications/wine-explorer.desktop"
> cat << EOF > "$PREFIX/share/applications/wine-winecfg.desktop"
> [Desktop Entry]
> Name=Wine Configuration
> Comment=Interface to set wine parameters
> Name[ar]=إعدادات واين
> Comment[ar]=لوحة تحكم بمنظومة واين
> Exec=${wine_cmd} winecfg
> Icon=winecfg
> Terminal=false
> Type=Application
> StartupNotify=true
> StartupWMClass=winecfg.exe
> Categories=X-Wine;
> X-Desktop-File-Install-Version=0.26
> EOF
> chmod +x "$PREFIX/share/applications/wine-winecfg.desktop"
> cat << EOF > "$PREFIX/share/applications/wine-uninstaller.desktop"
> [Desktop Entry]
> Name=Wine Software Uninstaller
> Comment=Interface to uninstall software
> Name[ar]=مزيل التطبيقات من واين
> Comment[ar]=واجهة إزالة التّطبيقات من واين
> Exec=${wine_cmd} uninstaller
> Icon=msiexec
> Terminal=false
> Type=Application
> StartupNotify=true
> StartupWMClass=uninstaller.exe
> Categories=X-Wine;
> X-Desktop-File-Install-Version=0.26
> EOF
> chmod +x "$PREFIX/share/applications/wine-uninstaller.desktop"
> cat << EOF > "$PREFIX/share/applications/wine-regedit.desktop"
> [Desktop Entry]
> Name=Regedit
> Comment=Wine registry editor
> Name[ar]=محرر السّجل
> Comment[ar]=محرر لسجلات واين
> Exec=${wine_cmd} regedit
> Icon=regedit
> Terminal=false
> Type=Application
> StartupNotify=true
> StartupWMClass=regedit.exe
> Categories=X-Wine;
> X-Desktop-File-Install-Version=0.26
> EOF
> chmod +x "$PREFIX/share/applications/wine-regedit.desktop"
> cat << EOF > "$PREFIX/share/applications/wine-wineboot.desktop"
> [Desktop Entry]
> Name=Wine Boot
> Comment=Simulate restart
> Name[ar]=إقلاع واين
> Comment[ar]=محاكاة إعادة التشغيل لواين
> Exec=${wine_cmd} wineboot
> Icon=wine
> Terminal=false
> Type=Application
> Categories=X-Wine;
> X-Desktop-File-Install-Version=0.26
> EOF
> chmod +x "$PREFIX/share/applications/wine-wineboot.desktop"
> }
> 
> check_termux
> wine_shortcut
> save it into a script and run

the winetricks dint install and the termux-desktop installer stop.. when it happens it means its incomplete instalation 
tx11start does nothing.

gurachan, 2025-02-24:
> the winetricks dint install and the termux-desktop installer stop.. when it happens it means its incomplete instalation tx11start does nothing.

Same here 😭..

[image]

gurachan, 2025-02-24:
Nvm it works now there seems new commit push related to it.. <3

sabamdarif, 2025-02-24:
i

sabamdarif, 2025-02-24:
i fount the real issue it will be fixed now for sure, the link was wrong i didn't notice that and for for winetricks termux remove that from their repo

ryangp, 2025-02-24:
> i found the real issue now, it will be fixed now for sure, the link was wrong i didn't notice that and for winetricks, termux remove that from their repo

wine file , win configuration, everythinh relaed to wine does not open

sabamdarif, 2025-02-25:
> > i found the real issue now, it will be fixed now for sure, the link was wrong i didn't notice that and for winetricks, termux remove that from their repo
> 
> wine file , win configuration, everythinh relaed to wine does not open

run 
```bash
wine winecfg
```
and tell me what happened

sabamdarif, 2025-02-25:
@gurachan try it fixed now also meke sure to use the latest installer (must)

ryangp, 2025-02-25:
//

[image]

> > > i found the real issue now, it will be fixed now for sure, the link was wrong i didn't notice that and for winetricks, termux remove that from their repo
> > 
> > 
> > wine file , win configuration, everythinh relaed to wine does not open
> 
> run
> 
> wine winecfg
> and tell me what happened

sabamdarif, 2025-02-25:
@ryangp  you issue isn't related to this create a new one, this was only for desktop entries which fixed now and i will close it

closed 2025-02-25 by sabamdarif

gurachan, 2025-02-28:
> i found the real issue now, it will be fixed now for sure, the link was wrong i didn't notice that and for winetricks, termux remove that from their repo

Fresh install .. still the same.. i dont know why it was labeled the best :'( mobox games open right of the box.. hangover nothing open.. the file manager opens tho..

sabamdarif, 2025-02-28:
it's label as best because in for most the people tested it mobox always giving them some issue

commit 6fc92bbb (2025-02-23): improve generic with hardware acceleration option, fix #77