sabamdarif/termux-desktop #78: It's getting stuck within a few moments of running.

NoReplyUI5, 2025-02-14:


sabamdarif, 2025-02-14:
launch Chromium from terminal and tell me what it shows
and what are the other issues you have ?

NoReplyUI5, 2025-02-14:
> launch Chromium from terminal and tell me what it shows
> and what are the other issues you have ?

I fixed that problem by reinstalling it. But there is one problem here which is the GUI getting stuck.

sabamdarif, 2025-02-15:
which install option you select, generic or custom ?

NoReplyUI5, 2025-02-15:
> which install option you select, generic or custom ?

I installed custom because I installed the same configuration last time.

Should I try generics for better performance?

sabamdarif, 2025-02-15:
no custom is better

sabamdarif, 2025-02-15:
which option you select for hardware acceleration

NoReplyUI5, 2025-02-15:
> which option you select for hardware acceleration 

[image]

sabamdarif, 2025-02-15:
which gpu you have, the gpu, and those  vulkan and opengl shouldn't be blank

sabamdarif, 2025-02-15:
ok i will push an update soon reinstall after that

NoReplyUI5, 2025-02-15:
[image]

> which gpu you have, the gpu, and those  vulkan and opengl shouldn't be blank

🐧

NoReplyUI5, 2025-02-15:
> ok i will push an update soon reinstall after that 

okay. i am confused 🤔

sabamdarif, 2025-02-15:
i mean i find the issue why it can't detect the gpu and i am fixing that right now

sabamdarif, 2025-02-15:
it's done you can try now

closed 2025-02-15 by sabamdarif