sabamdarif/termux-desktop #79: Termux x11 black screen

Anmol6002, 2025-02-15:
When launching it with tx11start it is showing just a cross instead of mouse cursor and a black screen but when starting with tx11start -nogpu runs and how to run hangover it's installed and is not same as mobox there's nothing happened when I run hangover just like mobox to run mobox .
Will GPU acceleration still work if I run vkmark vkcube it's showing the tests and gpu name adreno 613 but I am not sure if any application will work and how to launch application with icdwrapper with mesa i installed this one 5 option

sabamdarif, 2025-02-15:
> When launching it with tx11start it is showing just a cross instead of mouse cursor and a black screen but when starting with tx11start -nogpu runs

this mean two things first try running it other option like `tx11start --legacy` `tx11start --nodbus` if this work then great, otherwise it's gpu driver issue where try all 5 option from the opengl driver selection option there is i can do just try and see which work better for you

and if the tests showing gpu name right then it most likely that apps will also use the gpu

and for icd-wrapper, if during setup you chose icd-wrapper then it will automatically be used when you launch the desktop

Anmol6002, 2025-02-15:
> this mean two things first try running it other option like `tx11start --legacy` `tx11start --nodbus` if this work then great, otherwise it's gpu driver issue where try all 5 option from the opengl [...]
Tried both the options legacy and dbus it's not working going to reinstall

sabamdarif, 2025-02-15:
i think i should close it because there is nothing more can be done about it

closed 2025-02-15 by sabamdarif