sabamdarif/termux-desktop #80: Request for an Automated Installation Script for Termux Desktop

shiva1485, 2025-02-22:
Hey, I just wanted to say that your repo is amazing! I've been using it a lot on my devices, and I really appreciate the frequent updates and improvements.

I'm currently trying to install your Termux Desktop manually, but I'd love it if you could create an automated installation script that selects all the options automatically and installs everything without requiring user input. This way, it would be much easier to set up without needing to go through the prompts every time.

I previously tried making one myself, but since you update the repo frequently (which is great!), my script quickly became outdated. It would be awesome if you could maintain an automated version alongside the main installation script, so it always stays up to date.

Here’s the script I made for the latest version of your setup:
```
# Installing Termux Desktop Interface
curl -Lf https://raw.githubusercontent.com/sabamdarif/termux-desktop/main/setup-termux-desktop -o setup-termux-desktop
chmod +x setup-termux-desktop

# Running the setup script with predefined answers
echo -e " 
\n3  # Custom config hardware: Desktop
\n1  # Desktop environment: XFCE
\n1  # Theme: Basic
\n3  # Browsers: Chrome and Firefox
\n3  # Code Editors: VS Code and Geany
\n1  # Media Player: VLC
\n1  # Image Editor: GIMP
\n3  # Windows Compatibility: Wine Hangover
\nn  # No hardware acceleration
\ny  # Add wallpapers
\ny  # Configure Zsh shell
\ny  # Install terminal utilities
\ny  # Configure file manager
\n2  # Display Server: X11 & VNC
\ny  # Start Termux Desktop at startup
\n2  # Default connection method: VNC
\ny  # Add Linux container
\n1  # Linux Distro: Debian
\ny  # Add audio support
\ny  # Configure Zsh in Linux
\ny  # Install terminal utilities in Linux
\ny  # Create a user account
\n1  # User authentication: No password, only username
\nsefgh  # Default username
\ny  # Confirm username
\n   # Press enter to continue
\ny  # Modify storage permissions
" | ./setup-termux-desktop

```
This script is based on the latest version of your setup, and it selects all the necessary options automatically. I’d really appreciate it if you could create and maintain an official automated script alongside the main Termux Desktop script.

Thanks again for your hard work on this project—your repo has been incredibly useful! Looking forward to your thoughts. 🚀

sabamdarif, 2025-02-25:
yes i have a same kind of idea too, there just install once with the configuration you want, then copy /data/data/com.termux/files/usr/etc/termux-desktop/configuration.conf file to somewhere else so the next time all you have to do setup-termux-desktop --local-config /path/to/configuration.conf and then it will setup that old configuration and you have do nothing no prompt or anything like that, just run that command  wait and use

sabamdarif, 2025-02-25:
but i can't do it because of a small bug i have which i can fix, 
see this line https://github.com/sabamdarif/termux-desktop/blob/main/setup-termux-desktop#L652 
so it should print like this 
```
de_startup="xfce4-session"
de_name="xfce"
themes_folder="/data/data/com.termux/files/home/.themes"
icons_folder="/data/data/com.termux/files/home/.icons"
style_answer="1"
style_name="Basic Style"
```
but instead it do 
```
de_startup=xfce4-session
de_name=xfce
themes_folder=/data/data/com.termux/files/home/.themes
icons_folder=/data/data/com.termux/files/home/.icons
style_answer=1
style_name=Basic Style
```
so i have issue loading style name value. which is I don't k now why, and also yes it can be bypass after loading the value load the name based on style_answer value, but i don't want to do that. If it get fixed then i will start working on that.

sabamdarif, 2025-03-03:
add the option `--local-config /path/to/configuration.conf` do this and it will automatically setup the desktop with your configuration and it will not ask any confirmation

closed 2025-03-03 by sabamdarif

shiva1485, 2025-03-05:
> add the option `--local-config /path/to/configuration.conf` do this and it will automatically setup the desktop with your configuration and it will not ask any confirmation

🫡