sabamdarif/termux-desktop #84: Black Screen On Vnc Server

solakcemalettin, 2025-03-01:
[image]

sabamdarif, 2025-03-01:
run `vncstart --nogpu` and see what happened

hackingarjun, 2025-03-26:
Please Bro , fix this quickly 😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭

sabamdarif, 2025-03-26:
@hackingarjun did you tried launching it with `vncstart --nogpu` ?

hackingarjun, 2025-03-26:
Yes, but there is showing " vncstart not found " .

sabamdarif, 2025-03-26:
did you select vnc ( mean both option) during the setup ?

hackingarjun, 2025-03-26:
I select " without hardware acceleration " > then X11 > then all process
was automated. But I don't select vnc directly.

sabamdarif, 2025-03-26:
then how did you setup vnc and how did you launch it ?

hackingarjun, 2025-03-26:
I don't know what happened . I only selected " without hardware
acceleration " --> then x11 --> then select some options ( like username
etc.) --> then all process was automated , But I don't select vnc directly.

hackingarjun, 2025-03-26:
I don't know what happened . I only selected " without hardware
acceleration " --> then XFCE --> then select some options ( like username
etc.) --> then all process was automated , But I don't select vnc directly.

sabamdarif, 2025-03-26:
> Please Bro , fix this quickly 😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭

so what did you mean by this ?

hackingarjun, 2025-03-26:
Black Screen problem

sabamdarif, 2025-03-26:
in termux x11 then i think ?

Foum02, 2025-03-27:
This issue is in both Vnc and Termux:x11

sabamdarif, 2025-03-27:
there there could be two reason 
1st change hardware api
2nd run `setup-termux-desktop --change display` and chose 1 
then start termux x11 / vnc again

Foum02, 2025-03-27:
Method 2 did not work

hackingarjun, 2025-03-27:
Hye Bro 🙂 , termux desktop are now fully installed in my phone .
Actually my Termux was installed from F-droid . Then I'm installing Termux
from github and now it is working. Termux desktop have not work on F-droid
Termux .

Foum02, 2025-03-27:
So I have termux from GitHub

Foum02, 2025-03-27:
So I kind of installed Termux from GitHub, now I'll try to reinstall everything.

sabamdarif, 2025-03-27:
> from github and now it is working. Termux desktop have not work on F-droid

that's odd

Foum02, 2025-03-27:
Very strange, nothing works for me

hackingarjun, 2025-03-28:
What's problem ?

Foum02, 2025-03-28:
I think I realized what the problem was. The problem is with Termux:x11 or with Termux itself

hackingarjun, 2025-03-28:
Install all echosystem of Termux from github , that is solution .

sabamdarif, 2025-03-28:
for those who have black screen issue on termux x11 
i find out it was termux x11 for some device and it get fixed yesterday so download the new build
https://github.com/termux/termux-x11/releases/tag/nightly

MINEOGO, 2025-04-04:
btw I don't use "vncstart" or "tx11" i use "vncserver" and "vncserver :{port}" to start my vnc.

sabamdarif, 2025-04-04:
> btw I don't use "vncstart" or "tx11" i use "vncserver" and "vncserver :{port}" to start my vnc.

vncstart command also uses the vncserver command to start vnc

MINEOGO, 2025-04-04:
it doesn't work.. i mean i cannot see what port is the vnc being hosted on

coad2006, 2025-04-05:
Hii theee

MINEOGO, 2025-04-06:
try reinstalling ig

padz24, 2025-08-01:
> run `vncstart --nogpu` and see what happened

╭─ [padil   localhost] [~]
╰─❯ vncstart --nogpu                                       No command vncstart found, did you mean:
 Command vnstat in package vnstat

help me

hiptodude3, 2025-10-03:
This issue happens in x11 for some odd reason, I can't really get this working with my gpu, what may be the reason?