sabamdarif/termux-desktop #86: GUI Sticking issue

NoReplyUI5, 2025-03-04:
https://github.com/NoReplyUI5/repo-for-what-/raw/refs/heads/main/Record_2025-03-04-18-28-10.mp4

sabamdarif, 2025-03-04:
what happened in termux x11 ?

sabamdarif, 2025-03-04:
yes i do use discord but you it will be much easier if you join the telegram group do you use telegram ?

NoReplyUI5, 2025-03-04:
> yes i do use discord but you it will be much easier if you join the telegram group do you use telegram ?

yup i do 😇
Can you give me the link?

NoReplyUI5, 2025-03-04:
> what happened in termux x11 ?

i don't use it 🥹
It's difficult for me to use.

sabamdarif, 2025-03-04:
https://t.me/hello_android_0 -group
discord username @sabamdarif but i don't use discoed that much

sabamdarif, 2025-03-04:
> > what happened in termux x11 ?
> 
> i don't use it 🥹 It's difficult for me to use. I use VNC Viewer

try just once and see what happened just install the x11 app and `tx11start`, that's all it will launch using termux x11

NoReplyUI5, 2025-03-04:
> > > what happened in termux x11 ?
> > 
> > i don't use it 🥹 It's difficult for me to use. I use VNC Viewer
> 
> try just once and see what happened just install the x11 app and `tx11start`, that's all it will launch using termux x11

Okay, I'll try it and let you know today (afternoon {maybe before 2:00:PM - IST}). I sent you a friend request on Discord with the name **_ユンイン_** And I've also joined the Telegram group.

NoReplyUI5, 2025-03-04:
Nvm, I tried it now.

[image]

[image]

> Okay, I'll try it and let you know today (afternoon {maybe before 2:00:PM - IST}). I sent you a friend request on Discord with the name **_ユンイン_** And I've also joined the Telegram group.

sabamdarif, 2025-03-05:
you can ignore the second issue, can you once try `tx11start --nogpu` and `tx11start --nodbus --nogpu`

closed 2025-03-07 by NoReplyUI5