sabamdarif/termux-desktop #87: Apps are not showing up on whisker menu

Vishwajith-S-VIT, 2025-03-07:
I recently lost my termux data so I decided to re-install this script and when I opened the GUI and clicked the whisker menu/apps menu, it shows nothing, absolutely nothing, but files/usr/share/applications/ contains all the files though. What should be done to fix this? This error has occurred in xfce4 all modes.

Vishwajith-S-VIT, 2025-03-07:
[image]

Here is a screenshot for reference, it shows that I have clicked but nothing is opening, please help.

sabamdarif, 2025-03-08:
by the screenshot it look like the apps file are all corrupted, after data lose did you do a clean install ?

sabamdarif, 2025-03-08:
i found it it's a bug from termux https://github.com/termux/termux-packages/issues/23672 wait for them to fix it

Vishwajith-S-VIT, 2025-03-08:
> by the screenshot it look like the apps file are all corrupted, after data lose did you do a clean install ? 

Yes, I tried a reinstallation and saw that the issue was only with XFCE. So I'll wait for a while for termux packages to fix it then. Thanks for your help.

sabamdarif, 2025-03-08:
or for now if you want you can fix it just by installing this old version of this package
https://github.com/termux/termux-packages/actions/runs/13484892594

closed 2025-03-08 by sabamdarif