sabamdarif/termux-desktop #90: `autostart` error

filmnt, 2025-03-10:
[image]

sabamdarif, 2025-03-11:
now update the installer by running `setup-termux-desktop --update`
then try again, and let me know if it's fixed or not

filmnt, 2025-03-11:
> now update the installer by running `setup-termux-desktop --update` then try again, and let me know if it's fixed or not

`setup-termux-desktop --update` needs `&& cd .. && cd usr/bin && ./setup-termux-desktop`and updates work.

[image]

`setup-termux-desktop --change autostart` works. But nothing happens

sabamdarif, 2025-03-12:
for Appstore termux just pushed a update on pygobject yesterday and now pyhon isn't detecting the gtk libraries wait for them to fix it, i asked them to fix that, when that happens i will tell you that here

sabamdarif, 2025-03-12:
and for change autostart issue do this
`cd ~`
`cat .zshrc `and send me screenshot or the whole text from top to bottom

filmnt, 2025-03-12:
> and for change autostart issue do this `cd ~` `cat .zshrc `and send me screenshot or the whole text from top to bottom

[image]
[image]

sabamdarif, 2025-03-12:
do you want to enable autostart right ?
and if yes and happened after you type y

filmnt, 2025-03-12:
> do you want to enable autostart right ? and if yes and happened after you type y

yes, several times. but nothing happens

sabamdarif, 2025-03-12:
now update the installer `setup-termux-desktop --update` and try again

filmnt, 2025-03-12:
> now update the installer `setup-termux-desktop --update` and try again

[image]
[image]

just updated. but can't autostart

filmnt, 2025-03-12:
> just updated. but can't autostart

`Custom` `installation` - `autostart` isn't working either

sabamdarif, 2025-03-12:
type `tdconfig` and send what it shows

filmnt, 2025-03-12:
> type `tdconfig` and send what it shows

Phone

[image]

Tablet

[image]

sabamdarif, 2025-03-12:
now try once more i tried my self and it work just fine just update the installer and try again don't need to reinstall

filmnt, 2025-03-12:
> now try once more i tried my self and it work just fine just update the installer and try again don't need to reinstall

update and reinstall work fine. thx!

closed 2025-03-12 by sabamdarif