sabamdarif/termux-desktop #91: `pdrun` with `Obsidian`

filmnt, 2025-03-13:
`pdrun` doesn't work with `Obsidian` after latest update

it works on `debian`

[image]

sabamdarif, 2025-03-14:
if you are inside debian then the command will be `/opt/AppImageLauncher/Obsidian/obsidian --no-sandbox`
you said pdrun isn't working but i can't see anything on this photo send me the full screenshot when you run it with pdrun

filmnt, 2025-03-14:
> if you are inside debian then the command will be `/opt/AppImageLauncher/Obsidian/obsidian --no-sandbox` you said pdrun isn't working but i can't see anything on this photo send me the full screenshot when you run it with pdrun

[image]

sabamdarif, 2025-03-14:
do `cat $PREFIX/bin/pdrun` in termux and screenshot

filmnt, 2025-03-14:
> do `cat $PREFIX/bin/pdrun` in termux and screenshot

`pdrun /opt/~/obsidian --no-sandbox` is not working

[image]

sabamdarif, 2025-03-14:
sorry my bad replace all `$DISPLAY` to `$(echo $DISPLAY)`

sabamdarif, 2025-03-14:
it should fix that issue

filmnt, 2025-03-14:
> it should fix that issue

obsidian works and termux show this

[image]

closed 2025-03-14 by filmnt