sabamdarif/termux-desktop #92: Like 74 Else Error

JD3Ent, 2025-03-14:
Can you help me, I don't know what the issue is that I installed the Termux desktop for the first time with the generic settings, everything was fine, but I wanted to re install it with some custom settings. But Everytime I reinstall with custom settings and try to boot to desktop, I get a line 74 Else unexpected token syntax error. I tried to re install the generic build again and that still worked. This only happens whenever I do a custom build. Can you help me figure out what is going on? [image]

sabamdarif, 2025-03-14:
type `tdconfig` and send screenshot or copy the whole text and send that here

sabamdarif, 2025-03-14:
also do `cat $PREFIX/bin/tx11start` and send that too

JD3Ent, 2025-03-14:
[image]

[image]

sabamdarif, 2025-03-14:
send the full image of `cat $PREFIX/bin/tx11start`

JD3Ent, 2025-03-14:
> send the full image of `cat $PREFIX/bin/tx11start`

That is the full image, I screenshot Ted everything.

JD3Ent, 2025-03-14:
[image]

[image]

[image]

[image]

[image]

[image]

sabamdarif, 2025-03-14:
run this it will fix those errors 
`sed -i '/if ! \$use_nogpu; then/,/fi/ {/if \$use_debug; then/,/fi/d}' $PREFIX/bin/tx11start`

sabamdarif, 2025-03-14:
if it still didn't fix then in the tx11start file under `# Set environment variables if NOT using --nogpu`
remove the lines
```bash
if  $use_debug; then

else
      > /dev/null 2>&1 &
fi
```

sabamdarif, 2025-03-14:
> run this it will fix those errors `sed -i '/if ! \$use_nogpu; then/,/fi/ {/if \$use_debug; then/,/fi/d}' $PREFIX/bin/tx11start`

tell me if it's work

sabamdarif, 2025-03-14:
do `tx11start --nogpu`

sabamdarif, 2025-03-14:
you select the `6. The Vulkan-Icd-Wrapper Driver With Mesa` it don't work in most of the device as per what i seen so far it only work on latest samsung device only

sabamdarif, 2025-03-14:
for you i think the freedreno option will be best or try generic

JD3Ent, 2025-03-14:
Yeah I figured it was a gpu problem, I'll reinstall. The generic works fine everytime, but it doesn't come with wine installed or mobox so I can run windows games on my phone. Basically I just need the generic setup with mobox really (because mobox is the better windows emulation for mobile). So do you have a config file or can you tell me what I should select for the custom setup where everything is basically the same as generic but I have my GPU acceleration working and mobox installed?

sabamdarif, 2025-03-14:
for mobox after installing the desktop install mobox using it's own installer then start it one and stop it, after that run 
`wget -O "$PREFIX/bin/wine" "https://raw.githubusercontent.com/LinuxDroidMaster/Termux-Desktops/main/scripts/termux_native/mobox_run.sh"`
`chmod +x "$PREFIX/bin/wine"`
then start desktop and use the command `wine` to use wine

sabamdarif, 2025-03-14:
the if you don't know the generic setup uses the icd-wrapper and 2. OpenGL (VIRGL) and turnip

JD3Ent, 2025-03-14:
> the if you don't know the generic setup uses the icd-wrapper and 2. OpenGL (VIRGL) and turnip

Ahh, perfect, I'll try those things (including what you said for the generic setup you settings and then I'll let you know how it goes! Thank you, very helpful!

sabamdarif, 2025-03-16:
@JD3Ent did it fixed then i will close this issue

closed 2025-03-19 by sabamdarif