sabamdarif/termux-desktop #94: On Samsung galaxy s21 ultra, android 12 and android 14

CodeMasterCody3D, 2025-03-16:
I am on a working build on Feb 1st it could still work on a newer build, but on the newest build it's broken, it doesn't have hardware acceleration working properly for your newest build. So I resorted to the old build. Idk what has been modified but I want to update you on it. I'll come back with updates. I'm gonna re do the install of the newest, get the error and come back.

CodeMasterCody3D, 2025-03-16:
I see an error with the menu for the styling, I can't use hardware acceleration on proot.

[image]

[image]

[image]

CodeMasterCody3D, 2025-03-17:
I found that using ubuntu --root then logging in either with login cody then typing password or I can use su - cody to login and would show what I had in the screen shot from earlier. 

Now I found that I can log in with pdrun su - cody and it will show me logged intoo the proot with zsh correct. 

[image]

CodeMasterCody3D, 2025-03-17:
Never mind I found that using just, ubuntu I could log in as my user name. Sorry I need to read over all the documents.

sabamdarif, 2025-03-17:
that's good if you manage to fix that, by the way the `pdrun startxfce4` is wrong

sabamdarif, 2025-03-17:
and also are you on mali gpu ?

sabamdarif, 2025-03-17:
for zsh issue just run `ubuntu` to login into that distro and see what happened

CodeMasterCody3D, 2025-03-17:
> and also are you on mali gpu ?

I am on Adreno.

I use to use ubuntu --root 
Then I would type login
Then the user name
Then password. 
I am new to all of this and have gaps of knowledge missing.... I'll get better at understanding more and remembering things.

I made a script to install Orca arm64 version for me from my github.

Also when trying to install the freedreno drivers with the script it says it's not one of the selections.  So I have to manually install it as well and I made a script for that too.

https://github.com/CodeMasterCody3D/OrcaSlicer/releases/tag/arm64

CodeMasterCody3D, 2025-03-17:
I use the ubuntu to login now thank you. Also gpu hardware acceleration is working,  I just assumed it wasn't before due to the pdrun startxfce4 not working because I was installing Orca in a weird way and now that I don't need to do that and can use my scripts it's okay.

I am new and I'd like to ask if you could add the orca install to your appstore, what I need to do is fix my deb file tho and add all the required dependencies instead of relying on 

sudo apt-get install -f 

That will install missing dependencies with ubuntu 

If I run this on debian it just uninstall my Orca and doesn't install dependencies! Why tho? I can't find any answers online and maybe because my gaps in knowledge.

CodeMasterCody3D, 2025-03-17:
As of now only issues I see are

Typo for styling list
Can't install freedreno with selection in the installer says it's not a selection.

CodeMasterCody3D, 2025-03-17:
This is the file I am installing in proot to get hardware acceleration Graphics actually showing in OrcaSlicer. 

mesa-vulkan-kgsl_24.1.0-devel-20240120_arm64.deb

I have to install this after install of termux-desktop to gain 3d rendering for orca slicer. 

I'm using the icd-mesa-zink or what ever it's called in the installer with ternip.

sabamdarif, 2025-03-17:
i will add Orca Slicer in the app store and for freedreno let me see

sabamdarif, 2025-03-17:
> Also when trying to install the freedreno drivers with the script it says it's not one of the selections

can you send a pic of what it shows, i don't understand which part you are talking about

closed 2025-03-19 by sabamdarif

CodeMasterCody3D, 2025-03-19:
> > Also when trying to install the freedreno drivers with the script it says it's not one of the selections
> 
> can you send a pic of what it shows, i don't understand which part you are talking about

I will do a reset on my termux again and do a reinstall, I'll take a screen shot. Be back in 30 min or so.

CodeMasterCody3D, 2025-03-19:
You fixed it!! It's not doing that error now. Yes I also chose no to be able to choose freedreno before as well. I'm waiting now for installation.