sabamdarif/termux-desktop #95: Fork request

Ambientfish, 2025-03-18:
https://github.com/pelya/debian-noroot

Would you like to fork this repo/app, btw it's open-sourced so you will not face any issue 
The original app miss tons of adittions and optimizations so you can bring life to this debian container
The official app dev says that both opengl and Pulseaudio were broken, so you can add ur ultimate termux-desktop spice to the APP.

with love from morocco ;)

sabamdarif, 2025-03-18:
yes that good but i think the proot-distro is much better option for termux

Ambientfish, 2025-03-19:
Pelya said that the his app is also based on PRoot "THE ULTIMATE LINUX VIRTUALIZATION TOOL" so what? wdym with proot is better on termux 🤔
[image]

closed 2025-03-19 by sabamdarif