sabamdarif/termux-desktop #98: td.zsh-theme

1357op, 2025-03-19:
[image]
[image]

1357op, 2025-03-19:
After installation, the part that used to display the username as termux or ubuntu no longer works.

This issue likely occurred after adding the td.zsh theme.

sabamdarif, 2025-03-19:
you can fix it by manually adding this on that td.zsh-theme file

```bash
autoload -Uz colors && colors
autoload -Uz vcs_info

setopt PROMPT_SUBST

# Enable Git branch display
zstyle ':vcs_info:git:*' formats ' %F{magenta} %b%f'
precmd() { vcs_info }

# Exit status indicator
exit_status="%(?..%F{red}✘ %?%f)"

# Detect distro and set the correct icons
distro_id=$(grep '^ID=' /etc/os-release | cut -d= -f2 | tr -d '"' ) >/dev/null 2>&1
case "$distro_id" in
    kali) DISTRO_ICON="" ;;   # Kali Linux
    arch) DISTRO_ICON="" ;;   # Arch Linux
    ubuntu) DISTRO_ICON="" ;; # Ubuntu
    debian) DISTRO_ICON="" ;; # Debian
    fedora) DISTRO_ICON="" ;; # Fedora
    alpine) DISTRO_ICON="" ;; # Alpine
    void) DISTRO_ICON="" ;;   # Void Linux
    opensuse*|sles) DISTRO_ICON="" ;; # openSUSE
    gentoo) DISTRO_ICON="" ;; # Gentoo
    nixos) DISTRO_ICON="" ;; # NixOS
    *) DISTRO_ICON="" ;;      # Default Linux Icon
esac

# Symbols
FOLDER_ICON=""
PROMPT_SYMBOL="%F{green}❯%F{reset}"

# Check if running in Termux and set username accordingly
if [[ -n "$PREFIX" && "$PREFIX" == */com.termux/* ]]; then
    user_name="your user name "  # You can replace this with another method to get the preferred username
else
    user_name="$(whoami)"
fi

# User and host info inside a box with DISTRO_ICON in the middle
user_host="%F{blue}[%F{cyan}$user_name%F{yellow} $DISTRO_ICON %F{cyan}%m%F{blue}]%f"

# Current directory inside the same box
dir_display="%F{blue}[%F{yellow}\$FOLDER_ICON %F{yellow}%~%F{blue}]%f"

# Prompt layout (P10K-like)
PROMPT="
%B%F{green}╭─ $user_host $\dir_display\${vcs_info_msg_0_}
%B%F{green}╰─${PROMPT_SYMBOL} %F{reset}"
RPROMPT="$exit_status"
```

change your user name with your user name

closed 2025-03-20 by sabamdarif