sabamdarif/termux-desktop #99: Mobox or Wine Hangover?

TheRebo, 2025-03-20:
Hi!

Sorry for posting a question to the Issue...

I'm not familiar with Wine, so...
If I may know, what are the main differences between Mobox and Wine hangover? And in your opinion, which one is better?

sabamdarif, 2025-03-21:
both mobox and hangover design for same purpose, to use x86_64 windows software in arm device. mobox is design to run game much better and hangover it's the stock wine. So it's depend on what you want to do mobox for game and made by some other dev and hangover nothing specific and it's in termux repo

TheRebo, 2025-03-21:
> both mobox and hangover design for same purpose, to use x86_64 windows software in arm device. mobox is design to run game much better and hangover it's the stock wine. So it's depend on what you [...]

Ohh... Ok ok, thanks for the answer!

closed 2025-03-21 by TheRebo