sabamdarif/termux-desktop #105: permissom denied

MINEOGO, 2025-03-30:
well uhh help vncstart command doesn't exist

╭─ [mineogo  localhost] [~]
╰─❯ vncstart --help                                                    ✘ 148
zsh: permission denied: vncstart

sabamdarif, 2025-03-30:
generic install or custom install ?

MINEOGO, 2025-03-31:
generic install!

1357op, 2025-03-31:
> generic install!

During the custom installation process, select the X11 and VNC options.

sabamdarif, 2025-03-31:
type tdconfig and send ss

MINEOGO, 2025-03-31:
[image]

sabamdarif, 2025-03-31:
bro you didn't chose both (termux x11 + vnc), it's a generic install and vnc isn't included in generic setup

MINEOGO, 2025-03-31:
oh.. so do i reinstall?

sabamdarif, 2025-04-01:
do that and during the setup chose 3. custom, don't chose any of the generic option

MINEOGO, 2025-04-01:
ok thanks

closed 2025-04-02 by sabamdarif