sabamdarif/termux-desktop #106: Issue with Running Multiple Commands After ubuntu Command

1357op, 2025-03-30:
I am currently using the proot environment with Ubuntu. Before the update, when I typed the ubuntu command followed by additional commands, those commands were executed within the Ubuntu environment. However, after the update, the additional commands no longer run within the Ubuntu environment.

Could this issue be related to the line "source /data/data/com.termux/files/usr/etc/termux-desktop/common_functions" in "ubuntu" script? Currently, the common_functions file is missing, so I have commented out that line.

Is there any connection between this "common_functions" and the issue I'm facing?

sabamdarif, 2025-03-30:
the normal way is 1st run ubuntu, then it will  login into ubuntu after that run all the other commands one by one.
and by pasting 
>ubuntu
sudo apt install -y gcc libzmq5
curl -fsSL https://code-server.dev/install.sh | sh

did you mean paste all of them at once ?
and for common_functions you need that. otherwise the ubuntu <command> will not work properly.

and what is the error you are facing right now. can you send me what happening when you do that

1357op, 2025-03-31:
Yes, I wrote all the commands at once. Before this patch, executing them all at once worked fine within the Ubuntu environment. Although it’s not the normal way, it was a convenient method I used.

When I did a fresh install, I encountered an error while running Ubuntu, as shown in the image below. After commenting out the relevant line in the Ubuntu script, it started working properly again.

[image]

Here is an example of running all the commands at once:

[image]

sabamdarif, 2025-03-31:
```bash
#!/data/data/com.termux/files/usr/bin/bash

R="$(printf '\033[1;31m')"
G="$(printf '\033[1;32m')"
Y="$(printf '\033[1;33m')"
B="$(printf '\033[1;34m')"
C="$(printf '\033[1;36m')"
W="$(printf '\033[0m')"
BOLD="$(printf '\033[1m')"

termux_desktop_path="/data/data/com.termux/files/usr/etc/termux-desktop"
config_file="$termux_desktop_path/configuration.conf"
log_file="/data/data/com.termux/files/home/termux-desktop.log"
check_termux () 
{ 
    if [[ -z "$PREFIX" && "$PREFIX" != *"/com.termux/"* ]]; then
        echo "${R}[${R}☓${R}]${R}${BOLD}Please run it inside termux${W}";
        exit 0;
    fi
}
print_log () 
{ 
    local timestamp="$(date '+%Y-%m-%d %H:%M:%S')";
    local log_level="${2:-INFO}";
    local message="$1";
    echo "[${timestamp}] ${log_level}: ${message}" >> "$log_file"
}
print_success () 
{ 
    local msg;
    msg="$1";
    echo "${R}[${G}✓${R}]${G} $msg${W}";
    print_log "$msg"
}
print_failed () 
{ 
    local msg;
    msg="$1";
    echo "${R}[${R}☓${R}]${R} $msg${W}";
    print_log "$msg"
}
wait_for_keypress () 
{ 
    read -n1 -s -r -p "${R}[${C}-${R}]${G} Press any key to continue, CTRL+c to cancle...${W}";
    echo
}
check_and_create_directory () 
{ 
    if [[ ! -d "$1" ]]; then
        mkdir -p "$1";
        print_log "$1";
    fi
}
check_and_delete () 
{ 
    local file;
    local files_folders;
    for files_folders in "$@";
    do
        for file in $files_folders;
        do
            if [[ -e "$file" ]]; then
                if [[ -d "$file" ]]; then
                    rm -rf "$file" > /dev/null 2>&1;
                else
                    if [[ -f "$file" ]]; then
                        rm "$file" > /dev/null 2>&1;
                    fi;
                fi;
[... 442 lines cut ...]
    done
}
read_conf () 
{ 
    if [[ ! -f "$config_file" ]]; then
        print_failed " Configuration file $config_file not found";
        exit 0;
    fi;
    source "$config_file";
    print_success "Configuration variables loaded"
}
print_to_config () 
{ 
    local var_name="$1";
    local var_value="${2:-${!var_name}}";
    local IFS=' 	
';
    if grep -q "^${var_name}=" "$config_file" 2> /dev/null; then
        sed -i "s|^${var_name}=.*|${var_name}=${var_value}|" "$config_file";
    else
        echo "${var_name}=${var_value}" >> "$config_file";
    fi;
    print_log "$var_name $var_value"
}
```

sabamdarif, 2025-03-31:
here save it as coomon_function

sabamdarif, 2025-04-01:
you can do that, to that run
`nano $PREFIX/bin/ubuntu`
after 
```
elif [[ "$1" = "update" ]]; then
proot-distro login --user ......... sudo apt update
```
add a new line 
```
else
    proot-distro login --user "USERNAME" "ubuntu" --shared-tmp --termux-home -- env DISPLAY=:0 "$@"
```
then same and exit

the fine code should look like

````
elif [[ "$1" = "update" ]]; then
proot-distro login --user ......... sudo apt update
else
    proot-distro login --user "USERNAME" "ubuntu" --shared-tmp --termux-home -- env DISPLAY=:0 "$@"
fi
```

if you are confused about the `proot-distro login --user "USERNAME" "ubuntu" --shared-tmp --termux-home -- env DISPLAY=:0 ` line look at the line proot-distro login --user ......... sudo apt update you will understand

1357op, 2025-04-01:
Great it works!

"I think previously I used to press Enter after typing 'ubuntu' to continue writing 'the command' on the next line. 
Now, I will need to add a '' between 'ubuntu' and 'the command' on the next line.

Thank you for solving the issue!"

closed 2025-04-01 by 1357op