sabamdarif/termux-desktop #110: Ubuntu messages

1357op, 2025-04-05:
After clean install, what's that??

[image]

sabamdarif, 2025-04-05:
what is your system architecture ?

1357op, 2025-04-05:
System Info: 
System : Android 15 
Host: samsung SM-S938N 
Kernel : A Linux 6.6.30-android15-8- 30455426-abogkiS938NKSU1AYC3-4k 
CPU: QTI SM8750 (8 VCPU) 
Architectures: AARCH64 
Termux Version : >_ 0.119.0-beta.2-Github

sabamdarif, 2025-04-06:
run

```bash
getprop | grep abi
```

1357op, 2025-04-06:
`╭─ [dino  localhost] [~]
╰─❯ getprop | grep abi
[persist.device_config.aconfig_flags.connectivity.ip_reachability_ignore_never_reachable_neighbor_version]: [350820280]
[persist.device_config.aconfig_flags.connectivity.ip_reachability_mcast_resolicit_version]: [0]
[ro.product.cpu.abi]: [arm64-v8a]
[ro.product.cpu.abilist]: [arm64-v8a]
[ro.product.cpu.abilist32]: []
[ro.product.cpu.abilist64]: [arm64-v8a]
[ro.system.product.cpu.abilist]: [arm64-v8a]
[ro.system.product.cpu.abilist32]: []
[ro.system.product.cpu.abilist64]: [arm64-v8a]
[ro.vendor.product.cpu.abilist]: [arm64-v8a]
[ro.vendor.product.cpu.abilist32]: []
[ro.vendor.product.cpu.abilist64]: [arm64-v8a]

╭─ [dino  localhost] [~]
╰─❯`

1357op, 2025-04-06:
This never happened in previous termux desktop versions.

1357op, 2025-04-06:
Just type pd login, it works well...

```shell
╭─ [dino  localhost] [~] (ubuntu)
╰─❯ cd
sleep: error while loading shared libraries: libcutils.so: cannot open shared object file: No such file or directory
true: error while loading shared libraries: libcutils.so: cannot open shared object file: No such file or directory

╭─ [dino  localhost] [~]
╰─❯ exit

╭─ [dino  localhost] [~] (termux)
╰─❯ pd login ubuntu
Warning: CPU doesn't support 32-bit instructions, some software may not work.
root@localhost:~# cd
root@localhost:~# exit
logout

╭─ [dino  localhost] [~]
╰─❯
```

sabamdarif, 2025-04-06:
it's proot distro issue

try this, 1st run 
```
pd login ubuntu
```
then run 
```
apt update
apt install coreutils
```

sabamdarif, 2025-04-06:
after that run ubuntu to login into ubuntu

1357op, 2025-04-06:
Oh... its not works...🫠

```shell
╭─ [dino  localhost] [~] (termux)
╰─❯ pd login ubuntu
Warning: CPU doesn't support 32-bit instructions, some software may not work.
root@localhost:~# apt update
apt install coreutils
Hit:1 http://ports.ubuntu.com/ubuntu-ports noble InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports noble-updates InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports noble-security InRelease
Hit:4 https://ppa.launchpadcontent.net/mozillateam/firefox-next/ubuntu noble InRelease
Hit:5 https://ppa.launchpadcontent.net/mozillateam/thunderbird-next/ubuntu noble InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
coreutils is already the newest version (9.4-3ubuntu6).
The following packages were automatically installed and are no longer required:
  bubblewrap libfuse3-3 libpipewire-0.3-0t64
  libpipewire-0.3-common libspa-0.2-modules
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
root@localhost:~# exit
logout

╭─ [dino  localhost] [~]
╰─❯ ubuntu
Warning: CPU doesn't support 32-bit instructions, some software may not work.
sleep: error while loading shared libraries: libcutils.so: cannot open shared object file: No such file or directory
true: error while loading shared libraries: libcutils.so: cannot open shared object file: No such file or directory

╭─ [dino  localhost] [~] (ubuntu)
╰─❯ mkdir: error while loading shared libraries: libcutils.so: cannot open shared object file: No such file or directory
uname: error while loading shared libraries: libcutils.so: cannot open shared object file: No such file or directory
sleep: error while loading shared libraries: libcutils.so: cannot open shared object file: No such file or directory
true: error while loading shared libraries: libcutils.so: cannot open shared object file: No such file or directory
sleep: error while loading shared libraries: libcutils.so: cannot open shared object file: No such file or directory
true: error while loading shared libraries: libcutils.so: cannot open shared object file: No such file or directory
sleep: error while loading shared libraries: libcutils.so: cannot open shared object file: No such file or directory
true: error while loading shared libraries: libcutils.so: cannot open shared object file: No such file or directory
sleep: error while loading shared libraries: libcutils.so: cannot open shared object file: No such file or directory
true: error while loading shared libraries: libcutils.so: cannot open shared object file: No such file or directory
true: error while loading shared libraries: libcutils.so: cannot open shared object file: No such file or directory
sleep: error while loading shared libraries: libcutils.so: cannot open shared object file: No such file or directory

╭─ [dino  localhost] [~]
╰─❯ chsh -s /bin/bash
chsh: error while loading shared libraries: libcutils.so: cannot open shared object file: No such file or directory

╭─ [dino  localhost] [~]
╰─❯ exit                                 ✘ 127

╭─ [dino  localhost] [~]
╰─❯
```

1357op, 2025-04-06:
--reinstall coreutils & bash also not works...
What can I do?
Thanks for your time.

sabamdarif, 2025-04-06:
```
pd sh ubuntu
```
and in ubuntu run
```
chsh -s /bin/bash dino
```

sabamdarif, 2025-04-06:
this time it should work

1357op, 2025-04-06:
This is probably a zsh issue in the proot environment. 

If I change ubuntu to bash with that method, it works fine, but if I switch back to zsh, I get the same error. 

I hope the next update will fix this issue.

1357op, 2025-04-06:
Disabling the last line of .zshrc, "export LD_PRELOAD=/system/lib64/libskcodec.so", solved the problem. 

On my Samsung phone, the sound works fine in the desktop environment even when this is disabled. 

However, I have not confirmed whether the sound works properly in the proto environment.

[image]

[image]

[image]

commit a0b4af3c (2025-04-18): fix zsh theme issues

- fix auto cd isn't working
- fix .. alias for cd .. isn't working
- probably fix #110
- fix taking long time to login into proot-distro