sabamdarif/termux-desktop #111: setup-termux-desktop -c hw

1357op, 2025-04-12:
After clean setup, there is some error.

```shell
╭─ [dino  localhost] [~]
╰─❯ setup-termux-desktop -c hw
[✓] Configuration variables loaded
[-] Validating required configuration values...
[☓] The following required configuration values are not set:
  - distro_zsh_answer

╭─ [dino  localhost] [~]
╰─❯
``` 

What is distro_zsh_answer & how can I solve this?

sabamdarif, 2025-04-12:
sorry it's a mistake i made, to fix that do this
```
cd  /data/data/com.termux/files/usr/etc/termux-desktop
```
```
nano configuration.conf
```
and at the bottom add
```
distro_zsh_answer=y
```

i will fix it thatnk's

closed 2025-04-12 by sabamdarif

CryptoAsc, 2025-05-03:
╰─❯ setup-termux-desktop --change hw
[✓] Configuration variables loaded
[-] Validating required configuration values...
[☓] The following required configuration values are not set:
  - ext_wall_answer

sabamdarif, 2025-05-04:
`nano $PREFIX/etc/termux-desktop/configuration.conf` and add `ext_wall_answer=n`

mgueji, 2025-08-13:
Another one that happened to me trying to change the HW acceleration:
[☓] The following required configuration values are not set:
- user_name

I added user_name=myusername at the end and it is working

commit 34f31ba9 (2025-04-12): fix random no such file and directory error, fix #111 , add fnvim,fvim to search using fzf and open it on nvim/vim