sabamdarif/termux-desktop #114: GUI running problem

hacker50-us, 2025-04-22:
I am currently suffering from a problem that gui It does not work, so it gives a black screen with the mouse, and this is what gives in termux ```
/data/data/com.termux/files/usr/bin/tx11start: line 131: 23677 Aborted                 env DISPLAY=:0 XDG_CONFIG_DIRS=/data/data/com.termux/files/usr/etc/xdg dbus-launch --exit-with-session xfce4-session > /dev/null 2>&1

```
I hope that the problem will be solved quickly, especially since the problem may be related to hw

hacker50-us, 2025-04-22:
**New Updates**
I tried command
 ```
 tx11start --nogpu
```
And it worked with me desktop, I will try to experiment with all accelerator to see the error, and I will try glxgears every time even with nogpu 🤔

sabamdarif, 2025-04-22:
can you share your device info

hacker50-us, 2025-04-22:
> can you share your device info 

I used an old device (Huawei Y6s in this experiment) and this is the full information about it
```
Termux Variables:
TERMUX_API_VERSION=0.50.1
TERMUX_APK_RELEASE=GITHUB
TERMUX_APP_PACKAGE_MANAGER=apt
TERMUX_APP_PID=10352
TERMUX_APP__BUILD_DATA_DIR=/data/data/com.termux
TERMUX_APP__DATA_DIR=/data/user/0/com.termux
TERMUX_APP__LEGACY_DATA_DIR=/data/data/com.termux
TERMUX_APP__SE_FILE_CONTEXT=u:object_r:app_data_file:s0:c147,c271,c512,c768
TERMUX_APP__SE_INFO=default:targetSdkVersion=28:complete
TERMUX_IS_DEBUGGABLE_BUILD=1
TERMUX_MAIN_PACKAGE_FORMAT=debian
TERMUX_VERSION=0.118.2
TERMUX__HOME=/data/data/com.termux/files/home
TERMUX__PREFIX=/data/data/com.termux/files/usr
TERMUX__ROOTFS=/data/data/com.termux/files
TERMUX__SE_PROCESS_CONTEXT=u:r:untrusted_app:s0:c147,c271,c512,c768
TERMUX__USER_ID=0
Packages CPU architecture:
arm
Subscribed repositories:
# sources.list
deb http://mirror.mephi.ru/termux/termux-main stable main
# root-repo (sources.list.d/root.list)
deb http://mirror.mephi.ru/termux/termux-root root stable
# tur-repo (sources.list.d/tur.list)
deb https://tur.kcubeterm.com tur-packages tur tur-on-device tur-continuous
# x11-repo (sources.list.d/x11.list)
deb http://mirror.mephi.ru/termux/termux-x11 x11 main
Updatable packages:
All packages up to date
termux-tools version:
1.45.0
Android version:
9
Kernel build information:
Linux localhost 4.9.117+ #1 SMP PREEMPT Thu Dec 9 16:11:07 CST 2021 armv8l Android
Device manufacturer:
HUAWEI
Device model:
JAT-L29
Supported ABIs:
SUPPORTED_ABIS: armeabi-v7a,armeabi
SUPPORTED_32_BIT_ABIS: armeabi-v7a,armeabi
SUPPORTED_64_BIT_ABIS: 
LD Variables:
LD_LIBRARY_PATH=
LD_PRELOAD=/data/data/com.termux/files/usr/lib/libtermux-exec-ld-preload.so
Installed termux plugins:
com.termux.api versionCode:51
com.termux.x11 versionCode:15

```

And in talking about hw, I noticed that it worked for me only hw one and And in talking about hw, I noticed that it only worked for me hw one and he `5. OpenGL ES (ZINK VIRGL)` And I've really seen that lavapipe speeding up faster than speeding up  OpenGL ES (ZINK VIRGL) ~249% ( lavapipe was at ~118 fps and the second acceleration was only 34 fps) 
This must be the result of the many layers that translate to draw drawings,

sabamdarif, 2025-04-22:
the fps gap that can't be fixed only if the icd-wrapper works then it's great otherwise it will be like this in most of the cases, 
by the way which gpu you have

hacker50-us, 2025-04-22:
> the fps gap that can't be fixed only if the icd-wrapper works then it's great otherwise it will be like this in most of the cases, 
> by the way which gpu you have 

I have a very old CPU and it does not support most of the modern features (PowerVR GE 8320) AND IN THIS PROJECT, he always write me `UNKNOWN GPU `

sabamdarif, 2025-04-29:
Closing this issue as won't fix because because there is nothing can be done, just try all the hardware acceleration option and see whichever works for you.

closed 2025-04-29 by sabamdarif